KiCad/kicad-symbols #465: Datasheet defined twice?

arildj78, 2018-04-08:
It looks to me as if a link is supposed to be supplied to the component datasheet in two different places.

Symbol->Fields...->Datasheet->Field value
Symbol->Properties...->Description->Documentation File Name

Am I missing something or is this due to the recent redesign of the library?

jkriege2, 2018-04-08:
For the official libs we only use the latter option: http://kicad-pcb.org/libraries/klc/S6.2/

arildj78, 2018-04-08:
Thanks.
Is the `Datasheet field` and the `Show Datasheet` button in `Field Properties` a legacy function?

poeschlr, 2018-04-24:
Yes. As far as i am aware kicad 5 will fill the datasheet field from the dcm file when placing a symbol in eeschema (only if the field is empty).

More details see this mailing list conversation: https://www.mail-archive.com/kicad-developers@lists.launchpad.net/msg25829.html

I think this is the patch that got merged. (But there could be slight differences) https://www.mail-archive.com/kicad-developers@lists.launchpad.net/msg25887.html

closed 2018-04-24 by poeschlr