KiCad/kicad-symbols #466: Footprint filter not working as expected

arildj78, 2018-04-08:
I'm creating a new part (AD795) while following the guidelines in the KiCad Library Convention on [footprint filter](http://kicad-pcb.org/libraries/klc/S5.2/). Since I'm new here, I'm not sure what to expect from this setting, but as far as I can tell it does not do anything.

Just as a test I entered `DIP*` as the filter (totally wrong footprint btw) but I get access to all footprints anyway.

Is this to be expected or is it a bug?

Ratfink, 2018-04-08:
If I understand correctly, footprint filters only have any effect in CvPcb (Eeschema, Tools→Assign Footprints), or when placing symbols in Eeschema if the new option "Footprint previews in symbol chooser (experimental)" is enabled in Preferences→General Options→Display.

evanshultz, 2018-04-09:
@Ratfink is correct, as far as I know.

However, the footprint filter can be toggled in CvPcb. The 4 DIP-6 looking icons with green pins at the control the filtering. Here is a test project I had open showing filtering by pin count and keyword:
[image]

Here is my KiCad version: (5.0.0-rc2-dev-114-g8fcbb64a4), release build

If you are seeing all footprints in CvPcb when you're using filtering, that is an issue with the core KiCad program and you may want to file a bug at https://bugs.launchpad.net/kicad.

(Notice in the screenshot above that some switches are listed because they contain the text "DIP". It looks a bit weird, but it is correct.)

(Also, this symbol that I happened to have placed is built wrong because there is only one symbol for all packages but some had a thermal pad and others don't. That's why the DFN package isn't shown because all dimensionally-compatible footprints have 9 pins and this symbols erroneously only has 8 pins for the DFN package. Sorry for the aside, but I just noticed this and didn't want to show an example that would confuse further.)

poeschlr, 2018-04-09:
We could filter out the switches by using `*:DIP*` as filter. (KiCad 5 will have a slightly different behavior with regards to filtering then KiCad 4 had. Because it now allows filtering for libraries as well.)

Ratfink, 2018-04-18:
Closing this, since as far as I can tell there's no bug, just a misunderstanding about how a feature works.

poeschlr, 2018-04-18:
I think there is a problem here. And it is related with the bug i reported some time back over at the bugtracker. The answer from the developers: "Yes the behavior for filters changed. It should be fixed on the library side."

More detail: In kicad 4, footprint filters that start with some letter would mean that the footprint name needs to start with that letter. (In this case it meant that the footprint name must start with DIP for it to be shown.)
In kicad 5 this is no longer the case as kicad implicitly adds * at the beginning. This is such that filters work with the fact that library names are now included. (I agree that there is a bug in the software but the developers did not agree.) [For reference the bug report](https://bugs.launchpad.net/kicad/+bug/1751565)

---

So what needs to be done for fixing this? Well the footprint filters need to include *: at the beginning for them to behave the same as in kicad 4. I thought this will only affect filters with a single letter filter (R_*) but it seems it also affects other filters as well.

Ratfink, 2018-04-18:
@poeschlr I don't think that's what this issue was originally about: Arild saw *no* filtering, probably due to either having it disabled or searching with a part of KiCad that doesn't support filtering; what you're describing would make filters match too many footprints, but not *all of them*.  But if you want to use this issue to discuss that, alright.

But wait, if filters now have an implicit `*` at the beginning, couldn't we just prefix them all with `:`?  And if they have an implicit `*` at the beginning, do they also have one at the end, allowing us to remove the `*` from the end of all our filters?

poeschlr, 2018-04-18:
And what if the programmers in future remove the implicit `*`? Do we then add it back again? I think it helps to be specific.

I agree the best course of action would be to prefix all footprint filters with `*:`. Adding it only to filters where we for one reason or another find a problem will leave the possibility open that we miss problems.

Another reason for including them is that fp filters have an additional function to just telling the algorithm what to show. It also tells the user something about the symbol. 
In this case it communicates that the symbol does not care which lib the footprint comes from.
So i would keep the * at the front. (And at the end as well. Especially as i don't believe that there is a * added implicitly at the end.)

Ratfink, 2018-04-18:
Kind of a shame that we have to be defensive against a group we're supposed to be working together with, but *c'est la vie*.  I just tested and found that the `*` at the end is implicit now, but being explicit could prevent future trouble, so let's keep the `*` at the start and end.

How and when should we go about making this change?  It seems like it could easily break all open PRs.

poeschlr, 2018-04-18:
The more i think about this the more i become convinced that "fixing" this on the library level is the wrong approach. Implicitly adding * at the front and back of the footprint filters does not allow users to filter by prefix or suffix if they so desire. It does not even reach the goal of getting backwards compatibility for footprint filters that do not include a library filter.

For this reason i will again reach out to @stambaughw (via the original lauchnpad bug) in the hope that a better fix can be found.