KiCad/kicad-symbols #473: Specifying EP size in footprint filters?

evanshultz, 2018-04-11:
Since we started putting the EP size in the footprint names of applicable footprints (QFP, QFN, etc.), I've seen it be included in the footprint filters. This means the filters are valid for a suffix like `_ThermalVias` or without any suffix, but using the footprint filter will always only show footprints with a certain EP size.

Should the EP size be left off the footprint filter?

If the EP size is left off, the user is then shown a list of all parts matching in package type, XY size, and lead pitch. Any footprints matching those criteria would be shown in CvPcb, which would include variations with different EP sizes. If the user wanted to select an oversize EP for thermal reasons, then they could find and select it easily.

Of course there are drawbacks:
1. If the user selected a footprint with a smaller copper pad than the pad on the package it could possibly lead to poor thermals, unwanted solder bridging between the package pad and copper underneath if the mask was missing or scratched, etc.
2. If the EP size selected is much bigger than the copper pad, and the pin lengths of the package were longer than what the footprint was intended to use, the insides of the package pins could touch the copper pad and create a short. The paste applied on the copper pad would make a solder bridge.

So... I'm on the fence. I see possible positives and negatives. If a default footprint is given, which is think is overwhelmingly true, it seems like a slight net positive to me. But I'm not sure. Any other thoughts?

Ratfink, 2018-04-11:
I believe the only reason you gave for making this change is because it could potentially give improved thermal performance to have more copper.  Are there any other reasons to want a different EP size than the default?  If the only reason is for better thermal characteristics, I'm not sure why the same effect couldn't be achieved by adding copper zone fills and/or vias to the board in Pcbnew, without creating the potential for solder bridges you described.

evanshultz, 2018-04-12:
Yes, having a bigger EP was the reason why I thought of it. 2 PRs I looked at today were about the EP size and this popped into my head as I was reviewing.

I expect placing a footprint is much easier, and hopefully more safe, than creating a zone manually. But that is an option. Turning off footprint filtering in CvPcb means potentially trolling through a big list of footprints rather than a list that only varies by EP size.

Also, as I mentioned, I expect most or possibly all applicable symbols would have a default footprint. It wouldn't be required to pick a footprint at all if the user doesn't want to. But if the user wanted to select a different EP they'd have a curated list available in CvPcb already if footprint filtering is turned on.

No downside that I see and a small possible upside. The only effort is not putting the EP size in the footprint filter. I'm not super invested in this, but thought I'd bring it up to see the response.

poeschlr, 2018-04-19:
I agree that adding the ep size limits the usecases for the "footprint filter workflow". 
We set the correct footprint as the default anyways. Users who change the footprint should understand that this means they are responsible for checking if their selection makes sense.

We might want to add this information to the footprint filter section of the KLC.

evanshultz, 2018-04-19:
@poeschlr 
Do you mean you support removing the EP size from the footprint filter?

poeschlr, 2018-04-19:
Yes. 
In my opinion footprint filters should not include the ep size.  (Or any of the "modifier" fields like mask cutout size, pad size, ...)

---

By the way, the reason i assigned my self is such that i don't forget to add this to the KLC. it does not mean i will go over all symbols and make sure they do not include the EP size. 

I think most of them will not include it as we just recently added the ep size to the footprint names. Only the default footprints have been adjusted back then while fp filters stayed the same.

poeschlr, 2018-04-19:
Good point. 

Similarly symbols that use packages with missing pins should also include the missing pin specifier in their filter.

So i think the filters should then be `SOIC*1EP*...` and `SOIC*N9*...`.

evanshultz, 2018-06-30:
Decision made. Thanks all!

closed 2018-06-30 by evanshultz