KiCad/kicad-symbols #479: Alternate Node Sequence

arildj78, 2018-04-12:
This might be posted in the wrong place, if so, please point me in the right direction.
I can't figure out how this feature works.

[image]

I'm trying to change the pin mapping of a component when doing spice modeling, but I can't seem to get any change in the result when entering values in this field. Is this a bug or is it to be expected?

poeschlr, 2018-04-19:
It might be better to ask over at the [user forum](https://forum.kicad.info/) and if it is indeed a bug report it over at the [bugtracker](https://launchpad.net/kicad).

closed 2018-04-29 by poeschlr

evanshultz, 2018-08-08:
I'm finally getting around to playing with circuit simulation more, and figured it out. Then I discovered it was already at http://docs.kicad-pcb.org/stable/en/eeschema.html#simulator.

Anyway, it's from the SPICE model's perspective. So for the [OPA211 SPICE model](http://www.ti.com/lit/zip/sboc250) (see below), you would do this:
[image]

For reference, here is the header of the SPICE model:
```
* Copyright 2007 by Texas Instruments Corporation
* BEGIN MODEL OPA211
* MODEL FEATURES INCLUDE OUTPUT SWING, OUTPUT CURRENT THROUGH
* THE SUPPLY RAILS, RAIL-TO-RAIL OUTPUT STAGE, OUTPUT CURRENT
* LIMIT, OPEN LOOP GAIN AND PHASE WITH RL AND CL EFFECTS,
* SLEW RATE WITH TEMPERATURE EFFECTS, SETTLING TIME TO 0.01 %,
* OVERLOAD RECOVERY TIME, COMMON MODE REJECTION WITH FREQUENCY
* EFFECTS, OUTPUT IMPEDANCE, POWER SUPPLY REJECTION WITH
* FREQUENCY EFFECTS, INPUT VOLTAGE NOISE WITH 1/F AND FEATURE
* AT 20 MEGAHERTZ, INPUT CURRENT NOISE WITH 1/F, INPUT BIAS
* CURRENT, INPUT IMPEDANCE, INPUT COMMON MODE RANGE, INPUT
* OFFSET VOLTAGE WITH TEMPERATURE EFFECTS, AND QUIESCENT
* CURRENT VS VOLTAGE AND TEMPERATURE. MODEL INCLUDES INPUT
* PROTECTION DIODES.
* MODEL DOES NOT INCLUDE SHUTDOWN.
* PINOUT ORDER +IN -IN +V -V OUT
* PINOUT ORDER  3   2   7  4  6
.SUBCKT OPA211 3 2 7 4 6
```