KiCad/kicad-symbols #509: Naming for USB Type-C connector symbols

Ratfink, 2018-04-20:
As mentioned in https://github.com/KiCad/kicad-symbols/pull/310#issuecomment-379883615, our nomenclature for USB Type-C symbols could use improvement.  In particular, there are several types of connector defined in [the standard (106 MB .zip)](http://www.usb.org/developers/docs/usb_32_030818.zip) which aren't covered by the current naming: USB 2.0 Type-C plugs (with only 12 pins) and USB Type-C Power-Only plugs (with only 9 pins), as well as full-featured USB Type-C plugs with 24 pins.

Plugs and receptacles name a couple pins differently (CC1 and CC2 in the receptacle connect to CC and V<sub>CONN</sub> in the plug), so I would suggest keeping the current naming intact and extending it to support more types of connector by adding the number of populated pins.  Ones I know we would need, with checks for the ones we already have under a different name:

* [X] `USB_C_Receptacle-24`
* [ ] `USB_C_Plug-24`
* [X] `USB_C_Plug-22`
* [ ] `USB_C_Plug-12`
* [ ] `USB_C_Plug-9`

Another thing to discuss: the plugs with very few pins could have smaller symbols, or they could be kept the same size as the larger ones to act as drop-in replacements.  Which approach would be preferred?

evanshultz, 2018-04-20:
Which document(s) and page(s) are you looking at? I see this sort of stuff in `\usb_32_030818.zip\usb_32_030818\USB Legacy Cable and Connector\USB 3.1 Legacy Cable and Connector Revision 1.0.pdf` for non-C and `F:\_Evan\kicad\usb_32_030818.zip\usb_32_030818\USB Type-C\USB Type-C Specification Release 1.3.pdf` for type C. Are you looking elsewhere?

Ratfink, 2018-04-20:
That looks like the right PDF for the Type-C connector spec.  The pin names are shown on pages 21-22.

* Page 45 shows a Type-C receptacle with 16 pins for USB 2.0 only.
* Page 204 indicates that there are Type-C plugs with 24 pins.
* Page 48 shows the 12-pin USB 2.0 only Type-C plug.
* Pages 31-32 describe the 9-pin power-only Type-C plug.

evanshultz, 2018-04-25:
Yeah, it does seem to me like a unique symbol for the plug and receptacle are useful. Thanks for digging in and finding this!

- USB2.0 only: And page 46 shows a single-row version with only 12 pins; that would add `USB_C_Receptacle-12`. However, all of these are marked "informative". Pages 59-60 show the standard and USB 2.0-only pinouts; since USB 2.0-only is a subset they could share a symbol. Digi-Key only has https://media.digikey.com/pdf/Data%20Sheets/CNC%20Tech%20PDFs/C-AR20-AK510.pdf for a USB 2.0 Type C connector, and it has all 24 pins. So I'm thinking this is more of an electrical spec without any mechanical/physical changes.
- I don't quite get what you mean about page 204. I think 24 pins is standard. Can you clarify?
- Power-only: Since this is a depop with unused pins optional, I don't feel strongly we need another symbol just for it. But if we had it, that's fine with me too.

Overall, I think we're OK without adding any more symbols. Pages 122-123 give a nice overview of the electrical bits. If specific components are added that require more symbols then they should be added, but it seems to me our current symbols cover everything in the spec. Do you see any symbol that is truly missing or that we should support?

Also, this does mean we should put `Plug` and `Receptacle` in our footprint names, right? Otherwise we'll be missing an obvious chance to narrow down the available footprint list based on the symbol in CvPcb.

Lastly, I notice the spec shows (for example) `Dp1` and `SSRXn2`, with `p` and `n` and number at the end, while our symbols use `+` an `-` with numbers before the plus/minus symbol. That seems like something that could/should be fixed.

Ratfink, 2018-04-25:
For plugs, it's standard to omit the D+/- pins on the B side (B6 and B7), but some plugs do include them, as for direct-connect applications like USB sticks they can be re-mapped using alternate modes (as shown on page 204).  For an example of a 22-pin plug, see `Connector_USB:USB_C_Plug_Molex_105444` in the library.  For a 24-pin plug, see https://github.com/KiCad/kicad-footprints/pull/507.  Since real parts for both of these exist today, I think it would be useful to have symbols for both 22- and 24-pin plugs.

We do already have Plug and Receptacle in our footprint names for USB Type-C connectors.  I made sure of that when I created them. 🙂

The spec is inconsistent in its own naming of the pins.  The symbols we have follow the pin naming on pages 21-22.

evanshultz, 2018-04-30:
I still wonder if any more symbols are needed. The plug and receptacle are unique pinouts, so it makes sense to have those symbols to me, but otherwise if the user only needs fewer than the full number of pins why are more symbols needed? Aren't the pin numbers for each signal the same (so all footprints still work)? Can't the unneeded pins just not be used? Or are you thinking more about have one specific symbol for each use case and the user picks the one they want and then only sees the pins they will use?

Ratfink, 2018-04-30:
My main concern that makes me think more symbols are needed is footprint filtering.  The only way to support all use cases with a single plug symbol would be for it to have 24 pins (plus shield, of course).  This would make the symbol chooser's new footprint selection feature filter out 22-pin plugs, making them hard to discover.

Only having a 24-pin plug symbol would also mean that if a user wanted to use a 22-pin plug, there would be two pins in the schematic that simply shouldn't be there, because there's no corresponding pin in the connector or pad on the PCB.  This would confuse anyone checking a board against the schematic, and would also make Pcbnew complain every time the PCB is updated from the schematic, with errors like:

>Error: Component P1 pad B6 not found in footprint Connector_USB:USB_C_Plug_Molex_105444 
>Error: Component P1 pad B7 not found in footprint Connector_USB:USB_C_Plug_Molex_105444
>
>Total warnings: 0, errors: 2.
>Error: Errors occurred during the netlist update. Unless you fix them, your board will not be consistent with the schematics.

For all of these reasons, I think it's important to have symbols for depopulated USB Type-C connectors.  There is only an immediate need for 24- and 22-pin plugs and 24-pin receptacles, which means we only need one new symbol now.  They should all be renamed with the number of pins though, for consistency and future-proofing.