KiCad/kicad-symbols #510: Issues with Logic_LevelTranslator.lib

evanshultz, 2018-04-20:
From https://github.com/KiCad/kicad-symbols/pull/461

Checking symbol 'TXB0104RGY':
  Violating S3.6
    Pin name position offset
    Pin offset not preferred value
     - Pin offset (40) should be 20mils unless required by symbol geometry
  Violating S5.1
    For components with a single default footprint, footprint field is filled with valid footprint
    filename
    Footprint filter 'Texas*S*PVQFN*N14*ThermalVias*' does not match the footprint 'Package_DFN_QFN:Texas_S-PVQFN-N14' set for this symbol.
     - could not match 'Texas_S-PVQFN-N14' against filter 'Texas*S*PVQFN*N14*ThermalVias*'
     - could not match 'Package_DFN_QFN:Texas_S-PVQFN-N14' against filter 'Texas*S*PVQFN*N14*ThermalVias*'
Changed symbol 'Logic_LevelTranslator.lib:TXB0102YZP'
Checking symbol 'TXB0102YZP':
  Violating S3.6
    Pin name position offset
    Pin offset not preferred value
     - Pin offset (40) should be 20mils unless required by symbol geometry
  Violating S5.1
    For components with a single default footprint, footprint field is filled with valid footprint
    filename
    Footprint filter 'DSBGA?8*P0.5mm*' does not match the footprint 'Package_BGA:Texas_DSBGA-8_0.9x1.9mm_Layout2x4_P0.5mm' set for this symbol.
     - could not match 'Texas_DSBGA-8_0.9x1.9mm_Layout2x4_P0.5mm' against filter 'DSBGA?8*P0.5mm*'
     - could not match 'Package_BGA:Texas_DSBGA-8_0.9x1.9mm_Layout2x4_P0.5mm' against filter 'DSBGA?8*P0.5mm*'
Changed symbol 'Logic_LevelTranslator.lib:TXB0104PW'
Checking symbol 'TXB0104PW':
  Violating S3.6
    Pin name position offset
    Pin offset not preferred value
     - Pin offset (40) should be 20mils unless required by symbol geometry
Changed symbol 'Logic_LevelTranslator.lib:TXB0104ZXU'
Checking symbol 'TXB0104ZXU':
  Violating S3.6
    Pin name position offset
    Pin offset not preferred value
     - Pin offset (40) should be 20mils unless required by symbol geometry
  Violating S5.1
    For components with a single default footprint, footprint field is filled with valid footprint
    filename
    Footprint filter 'Texas*Junior*DSBGA*12*' does not match the footprint 'Package_BGA:Texas_MicroStar_Junior_BGA-12_2.0x2.5mm_Layout4x3_P0.5mm' set for this symbol.
     - could not match 'Texas_MicroStar_Junior_BGA-12_2.0x2.5mm_Layout4x3_P0.5mm' against filter 'Texas*Junior*DSBGA*12*'
     - could not match 'Package_BGA:Texas_MicroStar_Junior_BGA-12_2.0x2.5mm_Layout4x3_P0.5mm' against filter 'Texas*Junior*DSBGA*12*'
Changed symbol 'Logic_LevelTranslator.lib:TXB0104RUT'
Checking symbol 'TXB0104RUT':
  Violating S3.6
    Pin name position offset
    Pin offset not preferred value
     - Pin offset (40) should be 20mils unless required by symbol geometry
New symbol 'Logic_LevelTranslator.lib:FXMA108'
Changed symbol 'Logic_LevelTranslator.lib:TXB0108DQSR'
Checking symbol 'TXB0108DQSR':
  Violating S3.6
    Pin name position offset
    Pin offset not preferred value
     - Pin offset (40) should be 20mils unless required by symbol geometry
Changed symbol 'Logic_LevelTranslator.lib:TXB0104YZT'
Checking symbol 'TXB0104YZT':
  Violating S3.6
    Pin name position offset
    Pin offset not preferred value
     - Pin offset (40) should be 20mils unless required by symbol geometry
  Violating S5.1
    For components with a single default footprint, footprint field is filled with valid footprint
    filename
    Footprint filter 'DSBGA*12*' does not match the footprint 'Package_BGA:Texas_DSBGA-12_1.36x1.86mm_Layout3x4_P0.5mm' set for this symbol.
     - could not match 'Texas_DSBGA-12_1.36x1.86mm_Layout3x4_P0.5mm' against filter 'DSBGA*12*'
     - could not match 'Package_BGA:Texas_DSBGA-12_1.36x1.86mm_Layout3x4_P0.5mm' against filter 'DSBGA*12*'
Changed symbol 'Logic_LevelTranslator.lib:TXB0104D'
Checking symbol 'TXB0104D':
  Violating S3.6
    Pin name position offset
    Pin offset not preferred value
     - Pin offset (40) should be 20mils unless required by symbol geometry

HubertD, 2018-07-01:
working on this: https://github.com/KiCad/kicad-symbols/pull/720

evanshultz, 2018-07-03:
Lib now has no violations from Travis!

closed 2018-07-03 by evanshultz