KiCad/kicad-symbols #520: Invalid footprint associations in Amplifier_Audio.lib

Ratfink, 2018-04-23:
As Wayne reported [here](https://lists.launchpad.net/kicad-lib-committers/msg00547.html), there are some invalid footprint associations in Amplifier_Audio.lib which need to be fixed.  They're old references to `Package_TO_SOT_THT:TO-220-5_Vertical_StaggeredType1`, which need to be changed to `Package_TO_SOT_THT:TO-220-5_Vertical_StaggeredType1_Py3.7mm` assuming that is indeed the right Y pitch.

poeschlr, 2018-04-23:
I did not even know about that mailing list. I think i should join it then. (If i find out how.)

stambaughw, 2018-04-23:
Below is an updated list with all of the symbol libraries with broken
footprint links as of c3c0a08b28.  In some cases there were so many
broken footprint links that I just included the library name.  Any time
a footprint and/or footprint library are renamed, someone needs to a
search and replace of the changed footprint/library name to all of the
symbol libraries.  I know this is going to be an effort to fix all of
these but we should make an effort to get it done by the stable 5 release.

Battery_Management: LTC4007
Converter_DCDC: All
Diode_Bridge: ABS2, MBL104S
Driver_LED: LT3465
Driver_Motor: L298HN
FPGA_Actel: All
Interface_CAN_LIN: MCP2557FD-xMF, MCP2558FD-xMF
Interface_Optical: Almost all
Interfact_UART: ADM1491EBR
LED: CLV1L-FKB, LED_Cree_XHP50_12V, LED_Cree_XHP50_6V,
LED_Cree_XHP70_12V, LED_Cree_XHP70_6V
MCU_Cypress: CY7C68013A-56LTX, CY8C4xx7LQI-4xx, and many more.
MCU_Nordic: nRF52810-QCxx
MCU_ST_STM32F0: STM32F038E6Yx, STM32F038G6Ux, STM32F042C4Ux,
STM32F042G4Ux, and many more.
MCU_ST_STM32F1: STM32F100R4Hx
MCU_ST_STM32F2
MCU_ST_STM32F3
MCU_ST_STM32F4
MCU_ST_STM32F7
MCU_ST_STM32HY
MCU_ST_STM32L0
MCU_ST_STM32L1
MCU_ST_STM32L4
MCU_ST_STM32L4+
MCU_Texas_MSP430
Memory_Flash
Memory_RAM
Memory_ROM
Regulator_Linear: L200, LT3080xT
Regulator_Switching: KA5H0265RCTU, KA5H0265RCYDTU, LM2734Y, LM2734Y, and
many more.
Relay: G5Q-1, G5Q-1A, and many more.
RF: TR-52D
RF_Module: DWM1001, TD1205, TD1208
Sensor: BME280, BME680
Sensor_Current: ACS709xLFTR-6BB, ACS711xEXLT-15AB, ACS780xLRTR-050B, and
many more.
Sensor_Humidity: Si7020-A20
Sensor_Motion: L3G4200, L3G4200D, L3GD20, LIS2DH, and many more.
Sensor_Proximity: KRC011, TSSP58P38
Sensor_Temperature: KTY81, KTY83, KTY84
Transistor_BJT: 2SB631, 2SB817, 2SD1047, 2SD600, and many more.
Transistor_FET: IRF40DM229, IRF6718L2, IRF6802SD, IRF6892S, and many more.
Transistor_IGBT: IRG4PF50W
Triac_Thyristor: TIC106
Video: AD9895, AD9891

Ratfink, 2018-04-23:
With regards to the STM32 libraries, the missing footprints are known, with the issue https://github.com/KiCad/kicad-footprints/issues/374 tracking the progress.

poeschlr, 2018-04-23:
I assume some of the things will be fixed with https://github.com/KiCad/kicad-symbols/pull/455

poeschlr, 2018-04-24:
Reformatted the list by @stambaughw as a tasklist (not possible in e-mail replies as they do not support markdown.)

- [ ] Battery_Management: LTC4007
- [ ] Converter_DCDC: All
- [ ] Diode_Bridge: ABS2, MBL104S
- [ ] Driver_LED: LT3465
- [ ] Driver_Motor: L298HN
- [ ] FPGA_Actel: All
- [ ] Interface_CAN_LIN: MCP2557FD- xMF, MCP2558FD- xMF
- [ ] Interface_Optical: Almost all
- [ ] Interfact_UART: ADM1491EBR
- [ ] LED: CLV1L- FKB, LED_Cree_XHP50_12V, LED_Cree_XHP50_6V,
- [ ] LED_Cree_XHP70_12V, LED_Cree_XHP70_6V
- [ ] MCU_Cypress: CY7C68013A- 56LTX, CY8C4xx7LQI- 4xx, and many more.
- [ ] MCU_Nordic: nRF52810- QCxx
- [ ] MCU_ST_STM32F0: STM32F038E6Yx, STM32F038G6Ux, STM32F042C4Ux,
- [ ] STM32F042G4Ux, and many more.
- [ ] MCU_ST_STM32F1: STM32F100R4Hx
- [ ] MCU_ST_STM32F2
- [ ] MCU_ST_STM32F3
- [ ] MCU_ST_STM32F4
- [ ] MCU_ST_STM32F7
- [ ] MCU_ST_STM32HY
- [ ] MCU_ST_STM32L0
- [ ] MCU_ST_STM32L1
- [ ] MCU_ST_STM32L4
- [ ] MCU_ST_STM32L4+
- [ ] MCU_Texas_MSP430
- [ ] Memory_Flash
- [ ] Memory_RAM
- [ ] Memory_ROM
- [ ] Regulator_Linear: L200, LT3080xT
- [ ] Regulator_Switching: KA5H0265RCTU, KA5H0265RCYDTU, LM2734Y, LM2734Y, and
- [ ] many more.
- [ ] Relay: G5Q- 1, G5Q- 1A, and many more.
- [ ] RF: TR- 52D
- [ ] RF_Module: DWM1001, TD1205, TD1208
- [ ] Sensor: BME280, BME680
- [ ] Sensor_Current: ACS709xLFTR- 6BB, ACS711xEXLT- 15AB, ACS780xLRTR- 050B, and
- [ ] many more.
- [ ] Sensor_Humidity: Si7020- A20
- [ ] Sensor_Motion: L3G4200, L3G4200D, L3GD20, LIS2DH, and many more.
- [ ] Sensor_Proximity: KRC011, TSSP58P38
- [ ] Sensor_Temperature: KTY81, KTY83, KTY84
- [ ] Transistor_BJT: 2SB631, 2SB817, 2SD1047, 2SD600, and many more.
- [ ] Transistor_FET: IRF40DM229, IRF6718L2, IRF6802SD, IRF6892S, and many more.
- [ ] Transistor_IGBT: IRG4PF50W
- [ ] Triac_Thyristor: TIC106
- [ ] Video: AD9895, AD9891

poeschlr, 2018-04-24:
DC/DC lib has a lot of missing symbols. The few symbols that are there are now fixed with #533

evanshultz, 2018-05-31:
Also missing is `Package_LGA:HLGA-10_2.5x2.5mm_P0.6mm` used by `Sensor_Pressure:LPS25HB`. Added above.

alaut, 2018-07-31:
It appears to me that TLP291-4 and others in the __Isolators.lib__ library are associated with the wrong footprints. I'm not nearly experienced enough to make any reliable attempt to correct through these yet... though I plan to contribute sooner than later.

For example. I'm not sure if technically the footprint associated with LTV-847S is correct or within a certain tolerance. Datasheet indicated 9mm width, footprint appears to be 9.53mm. I'm inclined to believe I'm wrong here.

TLP291-4 is definitely wrong though by a lot unless I'm missing something. According to the [datasheet](http://bit.ly/2Oyhfaj), the pitch is 1.27mm not 2.54 as associated in _Package_DIP:SMDIP-16_W7.62mm_

Ratfink, 2018-07-31:
This issue is about symbols with *invalid* footprint associations, i.e. names that are in the wrong format or don't exist.  I have filed a separate issue, #792, for this problem.

myfreescalewebpage, 2019-01-17:
Hello @Ratfink 
Just reading this issue, is it up to date or need to be updated ?
Regarding to CHIP and CHIP Pro, I'm the author of the original PR (symbol+footprint) and the missing footprint have been pushed at https://github.com/KiCad/kicad-footprints/pull/1156.
Joel

Ratfink, 2019-01-17:
Oh gosh, I really don't know, I haven't been keeping up with the KiCad library scene for the last five months or so.  I would hope the other librarians have been updating this issue as needed, but you'd have to ask them.

PanRe, 2019-10-22:
It seems to the case that all the STM32 MCU footprint links are broken too. However, might be easy to fix since the name of the footprints, as it seems, only need to be changed from "Package_XYZ" to "Housings_XYZ".

chschlue, 2019-10-22:
@PanRe 
Which ones?
Is this a call from the past?

PanRe, 2019-10-22:
@chschlue
Call from the past?
I just installed kicad v5 and cloned the latest kicad libs. As far as I have seen, all footprint links of the STM32 MCUs are wrong. There exists no footprint library called Package_xyz... From where I have cloned only Housings_xyz exist. The footprints inside those libs, however, have the same names. Hence only the library name need to be changed.

chschlue, 2019-10-22:
Did you accidentally clone v4 FP repos?

poeschlr, 2019-10-23:
@PanRe My guess you had version 4 installed sometime in the past and still have the footprint library table setup with the version 4 libs via the github plugin: https://forum.kicad.info/t/i-had-kicad-4-installed-previosly-now-i-updated-to-v5-now-i-have-some-problems-with-the-library-setup/11932

PanRe, 2019-10-23:
@chschlue 
No i did clone the correct one.
 @poeschlr 
Thank you very much! That was exactly the case! I did not expect that since i deinstalled KiCad v4 prior to installing v5. Nevertheless, thank you very much for your help!

Thank you guys!