KiCad/kicad-symbols #550: Splitting up large symbols into units

evanshultz, 2018-04-26:
@poeschlr @jkriege2 @Ratfink 
We haven't done much splitting up of large symbols into multiple units that I see or can recall. How do we want to handle it?

See https://github.com/KiCad/kicad-symbols/pull/457 for an example of an active PR with a split symbol.

I personally like splitting for some cases floating in my mind, but there are a lot different facets to this. If it's allowed it seems we should have rules or at least a guidelines. How should they be split up: number of pins, size of symbol, etc.? Are there only certain type(s) of symbols that could be split? Should we allow split and non-split versions of the same symbol as two users might have opposing views?

whitequark, 2018-05-05:
Ping. It's been 9 days with no replies...

Ratfink, 2018-05-07:
There are already a few symbols like this in the library.  I think they're mainly in the CPU libraries, e.g. CPU_PowerPC:MPC8641D.

The way it's grouped in #457 seems okay to me, but I feel like the placement of power pins should be just like in any other symbol.  Also, I'm not sure if a 49-pin symbol is big enough to really need to be split, but I wouldn't say it has to be not split either.

As for making guidelines for splitting large symbols, gosh, I wouldn't know where to begin.  If I had any suggestions, I would have brought them up when I reworked the STM32 library.

evanshultz, 2018-06-06:
How about this:
```
- If a symbol does not fit within an A4 size page with power pins on top and bottom and signal pins on the sides, it may be split into multiple units.
- Each unit should have a clear and defined grouping of pins, for example by function (GPIO B, ACPI, etc.)
- All power pins are to be placed on the last unit unless they have a specific relation to another unit.
- Note that KiCad does not generate a warning if not all units are placed!
<provide an example symbol image>
```

https://github.com/KiCad/kicad-symbols/pull/457 is one I'm planning to merge as a split symbol unless somebody objects.

See https://github.com/KiCad/kicad-symbols/pull/622 for another example. It's just barely short enough to fit on an A4 page, but because of obvious pin groups it might also be a candidate to split.
[image]

Ratfink, 2018-06-08:
That's a reasonable-sounding set of rules you have.  I think this phrasing might be better for the third bullet, using RFC 2119 language:

> * All power pins **must** be placed on the last unit unless they have a specific relation to another unit, in which case they **may** be placed on that unit instead.

Also maybe it should more clearly state if this is intended to make the last unit a power unit, or just to tack the power pins onto whatever the last unit may be.

evanshultz, 2018-06-08:
Sounds good to me. Thanks for suggesting it! And good point to clarify the power unit. How about:
```
- All power pins, and only power pins, must be placed on the last unit unless they have a specific relation to another unit, in which case they should be placed on that unit instead.
```

Here is another interesting situation not covered above:
- Some parts have small symbols but it may be nice to split them anyway.
- https://github.com/KiCad/kicad-symbols/pull/655 was submited for a triple diode with split symbols.
- https://github.com/KiCad/kicad-symbols/pull/659 puts all diodes in one symbol.
- I provided single and split current sense transformer symbols at https://github.com/KiCad/kicad-symbols/pull/632 since it's common to draw power and control circuitry separately.