KiCad/kicad-symbols #590: Mechanical:Mounting_Hole filters are wrong

antoniovazquezblanco, 2018-05-10:
In the symbols lib the component is named `Mechanical:Mounting_Hole` while the footprints are located in a `MountingHole.pretty` lib. What name is preferred?

Also FPFilters in the symbol should be updated acordingly.

Ratfink, 2018-05-10:
I vote for `MountingHole`.

Also, KLC says that `?` in a FPFilter is supposed to match zero or one characters, but now it only matches exactly one character.  We either need to update a lot of filters or complain to the developers some more.

antoniovazquezblanco, 2018-05-11:
Although offtopic, why don't we use regex patterns in FPFilter? I would vote for any behaviour that changes FPFilters towards regex.

poeschlr, 2018-05-11:
@antoniovazquezblanco 
That is something you need to take up with the devs.

@Ratfink 
I don't remember a time when ? was zero or one. This is clearly a mistake in the KLC.

antoniovazquezblanco, 2018-05-11:
@poeschlr I am just querying if this desirable from your point of view.

poeschlr, 2018-05-14:
If the current easy filtering is still available. Most usecases do not need full regex support. (As long as your footprint naming convention is well thought out.) 
Regex is way to complicated for us to expect that every kicad user must learn it. (In reality kicad does convert the current filter into a regex construct behind the scenes. So it should be easy once you have a way of distinguishing the advanced filter method from the current one via the graphical interface.)

Ratfink, 2018-05-14:
For the last couple of days, I've been trying to think of a use-case where regex filtering would make some existing situation better.  I have yet come up with one.  @antoniovazquezblanco Do you have a specific application for regex filters in mind that I just haven't thought of?

poeschlr, 2018-05-14:
The footprint string syntax is not random. It is a reduced set of the [Unix shell pattern matching syntax](https://www.gnu.org/software/findutils/manual/html_node/find_html/Shell-Pattern-Matching.html)

(reduced because i think the [string] syntax is not supported. But i could be wrong.

antoniovazquezblanco, 2018-05-23:
Given that, full regex would not make it better. Sorry for going offtopic here :).

Either way, syntax should be clarified and documentation updated to match the real behaviour if it is not currently ok.

Thank you very much

antoniovazquezblanco, 2018-05-23:
https://github.com/KiCad/kicad-symbols/pull/612

closed 2018-07-13 by evanshultz