KiCad/kicad-symbols #595: Missing symbol for DPDT switch

gioele, 2018-05-13:
The symbol library shipped with Kicad 5 RC1 contains no symbol for DPDT switches.

There is a SW_DPDT_x2 _Switch, dual pole double throw, separate symbols_, but the normal classic symbol is missing.

An example of the normal symbol is <https://upload.wikimedia.org/wikipedia/commons/e/ec/DPDT-symbol.svg>.

poeschlr, 2018-05-13:
Why make an issue instead of a pull request?

gioele, 2018-05-13:
Because somebody may have neither the time nor the experience to solve an issue they have found. :)

poeschlr, 2018-05-13:
And why would this somebody expect others to solve the problem for him? You know we are also volunteers.

Creating symbols and footprints is a vital skill to learn for every electronics engineer.

Kicad does not claim to have a lib for all possible parts. So this is not an issue but a wish. I close this as we do not accept wishes to create symbols/footprints (as we otherwise would have thousands of wishes open within a short time period.)
The only valid reason for such an issue would be if the symbol in question did exist in a previous version of the official kicad library.

closed 2018-05-13 by poeschlr