KiCad/kicad-symbols #601: Where should DDS ICs go?

evanshultz, 2018-05-15:
@poeschlr @jkriege2 @Ratfink 
As mentioned at https://github.com/KiCad/kicad-symbols/pull/597, Interface probably isn't the best lib for these ICs. DAC might work, since they're essentially generating an analog signal digitally, but they're not a normal DAC where they accept an external digital input.

We have 3 of them so it's not a unique class of parts but it's also not enough to obviously justify a new library.

What do you think?

Ratfink, 2018-05-15:
I've never even heard of these devices before, so weight my opinion very lightly:

Maybe these could go in the MixedSignal library proposed in https://github.com/KiCad/kicad-symbols/issues/451#issuecomment-381023574?

jkriege2, 2018-05-15:
On RS-Components.com ... they have them  in Semiconductors|Oscillator,Timer...|DDS ... so I would propose `Timer_DDS.lib`.

lorem-ipsum, 2018-05-15:
I don't find mixed signals matching, also because I can not imagine what mixed signals should be except "everything which does something with analog and digital". For `DAC` I find them to specialised as they usually can only generate sinusoidal outputs. `Timer_DDS ` would probably make most sense.

evanshultz, 2018-10-01:
https://github.com/KiCad/kicad-symbols/pull/1021 is a submission of another DDS symbol. Two of them, in fact.

IMO "timer" isn't a good classification because while it does have a clock its main objective is not to do with timing. These parts turn a digital representation of a waveform into an analog one but include internal bits for a specific purpose (usually making a precise sine wave) so it's not converting a digital signal into an analog one.

Mouser:
Semiconductors > Data Converter ICs > Data Acquisition ADCs/DACs - Specialized

I'm OK with ultimately creating a new lib of "Interface_DDS" since we now have 4 symbols. (I assume the "DSS" above is a typo of "DDS".) But for now, piling them into "Interface.lib" with an issue to move them into a new lib when the time is right seems like a good way to go. Agree?

pgrudzinski, 2018-10-01:
The Interface designation is very confusing to me as I would never thought about looking for them in this library, For me Interface means translating one bus to other, and indeed almost all ICs in Interface.lib do just that. It is not clear what to what are DDS interfacing. Timer would be more accurate, as high resolution programmable clock divider is one of their applications. ADC is also ok, as they are translating digital word to analog value, which is frequency in this case.

evanshultz, 2018-10-01:
Yeah. I kinda gave in on my last comment; I noted in the first comment that Interface isn't good.

We can still merge them into Interface.lib for now, to maintain the library structure.

Fortunately for users of KiCad all libraries will be searched so this is mostly an organizational thing and users won't have to select and search in just a specific library at a time.

evanshultz, 2018-10-02:
Then we can continue to dump DDS ICs into Interface.lib for now and create DAC_DDS.lib for them once we can push breaking changes to the libraries. Anyone disagree?

myfreescalewebpage, 2019-01-15:
@evanshultz  I think we can close this issue if you agree ?

Joel

evanshultz, 2019-01-16:
The DDS symbols haven't moved. If the decision about where they go is made, why not mark this issue with a milestone so it gets done for 5.1 or 6 or whenever it's due?

myfreescalewebpage, 2019-01-16:
Sorry, too much quickly reviewed, yes probably v6 if it breaks libraries?

poeschlr, 2019-01-17:
How many symbols existing at the time of the v5.0.0 release would be affected?
How many symbols that have either been added since or are in the process of being added would benefit from splitting this one library?

myfreescalewebpage, 2019-01-17:
I think there are AD9834, AD9850, AD9851, AD9910, AD9951 at this moment and I haven't seen pending PR for such kind of device.