KiCad/kicad-symbols #648: Strange behaviour of the Cvpcb

Misca1234, 2018-06-03:
I am not sure if you have already seen this or already reported and I guess it is not a library issue really but a program one.

It seems that Cvpcb have difficulties with foot prints where the size of the pad is not 3 numbers

I have made a symbol ESP8089 which I associate with the "wrong" footprint 
(just to show that there is not an error with the symbol or so, the strange behaviour is described further down.)

[image]

[image]

[image]

Now, if I change to the more correct foot print, QFN-32-1EP_5x5mm_P0.5mm_EP3.6x3.6mm
I get the "wrong" behaviour

[image]

[image]

[image]

It does not longer "catch" anything with the foot print filter

Misca1234, 2018-06-03:
Something is very strange here, here is a screen shoot of the footprint, the pin number is way wrong

[image]

The only one using that footprint, is nRF52810-QCxx in MCU_Nordic.lib
Shall I fix this footprint in same push as the symbol I want to push using this foot print ?

evanshultz, 2018-06-04:
@Misca1234 
Good find. So this doesn't have anything to do with CvPcb.

Yes, please fix the footprint. It won't be in the same PR as the symbol, but a complementary PR would be nice.

@poeschlr @Ratfink 
This footprint was added at https://github.com/KiCad/kicad-footprints/pull/518. I suspect the pin array feature was used four times but the starting pin number wasn't incremented on each row of pins and instead left at `1`. Maybe it's best to show copper pads and pad numbers (not just for us but all contributors) so something like this can be caught at a glance? https://github.com/KiCad/kicad-footprints/pull/612 is another one where the copper wasn't enabled during the screenshot and without that info this kind of thing may be more likely.

Ratfink, 2018-06-04:
Aw gosh darnit, I merged that one!  I was probably expediting the review because of all the work to fix #520, got lazy, and didn't load it in KiCad before merging.  Since I'm responsible for that footprint being merged, I'll fix it.

Misca1234, 2018-06-04:
>So this doesn't have anything to do with CvPcb.

No, it seems that the pad numbering messed stuff up for the Cvpcb.

> I'll fix it.

Ok, then I wont do anything with it then

closed 2018-06-04 by Misca1234

poeschlr, 2018-06-04:
And i created it so... Wasn't my best hour i guess.