KiCad/kicad-symbols #650: Conn_Coaxial footprint filters need improvement

Ratfink, 2018-06-04:
[KLC]: http://kicad-pcb.org/libraries/klc/ "KiCad Library Convention"
[G1.1]: http://kicad-pcb.org/libraries/klc/G1.1/ "Only standard characters are used for naming libraries and components"
[G1.2]: http://kicad-pcb.org/libraries/klc/G1.2/ "Each library is limited to 250 items"
[G1.3]: http://kicad-pcb.org/libraries/klc/G1.3/ "Libraries are organized by functionality"
[G1.4]: http://kicad-pcb.org/libraries/klc/G1.4/ "English language should be used throughout libraries"
[G1.5]: http://kicad-pcb.org/libraries/klc/G1.5/ "Plural naming is to be avoided"
[G1.6]: http://kicad-pcb.org/libraries/klc/G1.6/ "Capitalization conventions"
[G1.7]: http://kicad-pcb.org/libraries/klc/G1.7/ "Library files use Unix style line endings"
[G1.8]: http://kicad-pcb.org/libraries/klc/G1.8/ "Components can only contain features supported in latest stable release"
[G1.9]: http://kicad-pcb.org/libraries/klc/G1.9/ "Dimensional units"
[G2.1]: http://kicad-pcb.org/libraries/klc/G2.1/ "Defining generic and atomic parts"
[S1.1]: http://kicad-pcb.org/libraries/klc/S1.1/ "Symbol libraries should be categorized by function"
[S2.1]: http://kicad-pcb.org/libraries/klc/S2.1/ "General symbol naming guidelines"
[S2.2]: http://kicad-pcb.org/libraries/klc/S2.2/ "Non-functional variations in part number should be replaced with wildcard"
[S2.3]: http://kicad-pcb.org/libraries/klc/S2.3/ "Where atomic parts are available in multiple footprint options, a separate symbol must be drawn for each footprint"
[S3.1]: http://kicad-pcb.org/libraries/klc/S3.1/ "Origin is centered on the middle of the symbol"
[S3.2]: http://kicad-pcb.org/libraries/klc/S3.2/ "Text fields should use a common text size of 50mils"
[S3.3]: http://kicad-pcb.org/libraries/klc/S3.3/ "Symbol outline and fill requirements"
[S3.4]: http://kicad-pcb.org/libraries/klc/S3.4/ "Symbols with complex functionality may incorporate simple functional diagrams"
[S3.5]: http://kicad-pcb.org/libraries/klc/S3.5/ "Pin connection points must be placed outside of the symbol"
[S3.6]: http://kicad-pcb.org/libraries/klc/S3.6/ "Pin name position offset"
[S3.7]: http://kicad-pcb.org/libraries/klc/S3.7/ "Pin numbering for exposed pads"
[S3.8]: http://kicad-pcb.org/libraries/klc/S3.8/ "Multi unit symbols"
[S4.1]: http://kicad-pcb.org/libraries/klc/S4.1/ "General pin requirements"
[S4.2]: http://kicad-pcb.org/libraries/klc/S4.2/ "Pins should be grouped by function"
[S4.3]: http://kicad-pcb.org/libraries/klc/S4.3/ "Rules for pin stacking"
[S4.4]: http://kicad-pcb.org/libraries/klc/S4.4/ "Pin electrical type"
[S4.5]: http://kicad-pcb.org/libraries/klc/S4.5/ "Pins not connected on the footprint may be omitted from the symbol"
[S4.6]: http://kicad-pcb.org/libraries/klc/S4.6/ "Hidden pins"
[S4.7]: http://kicad-pcb.org/libraries/klc/S4.7/ "Active low pins should be designated using a bar above the symbol name"
[S5.1]: http://kicad-pcb.org/libraries/klc/S5.1/ "Symbols with a default footprint link to a valid footprint file"
[S5.2]: http://kicad-pcb.org/libraries/klc/S5.2/ "Footprint filters should match all appropriate footprints"
[S6.1]: http://kicad-pcb.org/libraries/klc/S6.1/ "Component Reference Designator (RefDes)"
[S6.2]: http://kicad-pcb.org/libraries/klc/S6.2/ "Component fields must be filled appropriately"
[S6.3]: http://kicad-pcb.org/libraries/klc/S6.3/ "Component metadata is added to symbol (and all aliases)"
[S7.1]: http://kicad-pcb.org/libraries/klc/S7.1/ "Power flag symbols"
[S7.2]: http://kicad-pcb.org/libraries/klc/S7.2/ "Graphical symbols"
[F1.1]: http://kicad-pcb.org/libraries/klc/F1.1/ "Footprint libraries are categorized by function"
[F1.2]: http://kicad-pcb.org/libraries/klc/F1.2/ "Connector footprint libraries"
[F2.1]: http://kicad-pcb.org/libraries/klc/F2.1/ "General footprint naming conventions"
[F2.2]: http://kicad-pcb.org/libraries/klc/F2.2/ "Footprint naming field prefixes"
[F2.3]: http://kicad-pcb.org/libraries/klc/F2.3/ "Manufacturer specific version of generic footprints"
[F2.4]: http://kicad-pcb.org/libraries/klc/F2.4/ "Footprint naming for non-standard pin numbering"
[F2.5]: http://kicad-pcb.org/libraries/klc/F2.5/ "Footprint naming conventions for specific components"
[F3.1]: http://kicad-pcb.org/libraries/klc/F3.1/ "SMD chip package naming conventions"
[F3.2]: http://kicad-pcb.org/libraries/klc/F3.2/ "Resistor naming conventions"
[F3.3]: http://kicad-pcb.org/libraries/klc/F3.3/ "Capacitor naming conventions"
[F3.4]: http://kicad-pcb.org/libraries/klc/F3.4/ "SMD IC package naming conventions"
[F3.5]: http://kicad-pcb.org/libraries/klc/F3.5/ "THT IC package naming conventions"
[F3.6]: http://kicad-pcb.org/libraries/klc/F3.6/ "Connector naming conventions"
[F3.7]: http://kicad-pcb.org/libraries/klc/F3.7/ "Fuse naming conventions"
[F4.1]: http://kicad-pcb.org/libraries/klc/F4.1/ "Datasheet recommendations take priority"
[F4.2]: http://kicad-pcb.org/libraries/klc/F4.2/ "Pin 1 should be located at the top left"
[F4.3]: http://kicad-pcb.org/libraries/klc/F4.3/ "Connected copper elements have the same pad number"
[F4.4]: http://kicad-pcb.org/libraries/klc/F4.4/ "Thermal pads"
[F4.5]: http://kicad-pcb.org/libraries/klc/F4.5/ "Specifying footprint keepout areas"
[F4.6]: http://kicad-pcb.org/libraries/klc/F4.6/ "Local clearance and settings should be set to zero"
[F5.1]: http://kicad-pcb.org/libraries/klc/F5.1/ "Silkscreen layer requirements"
[F5.2]: http://kicad-pcb.org/libraries/klc/F5.2/ "Fabrication layer requirements"
[F5.3]: http://kicad-pcb.org/libraries/klc/F5.3/ "Courtyard layer requirements"
[F6.1]: http://kicad-pcb.org/libraries/klc/F6.1/ "Footprint placement type must be set to surface mount"
[F6.2]: http://kicad-pcb.org/libraries/klc/F6.2/ "Footprint anchor should be placed in the middle of the component body"
[F6.3]: http://kicad-pcb.org/libraries/klc/F6.3/ "Pad requirements for SMD footprints"
[F7.1]: http://kicad-pcb.org/libraries/klc/F7.1/ "Footprint placement type must be set to Through Hole"
[F7.2]: http://kicad-pcb.org/libraries/klc/F7.2/ "Footprint anchor should placed at the location of Pin-1"
[F7.3]: http://kicad-pcb.org/libraries/klc/F7.3/ "Pin 1 should be rectangular, and other pads circular or oval"
[F7.4]: http://kicad-pcb.org/libraries/klc/F7.4/ "Pad requirements for THT footprints"
[F7.5]: http://kicad-pcb.org/libraries/klc/F7.5/ "Minimum annular ring width"
[F7.6]: http://kicad-pcb.org/libraries/klc/F7.6/ "Minimum hole diameter"
[F8.1]: http://kicad-pcb.org/libraries/klc/F8.1/ "Virtual components"
[... 7 lines cut ...]
[M2.2]: http://kicad-pcb.org/libraries/klc/M2.2/ "Model alignment and scaling"

Our current footprint filters for the `Conn_Coaxial*` symbols are as follows:

    *BNC*
    *SMA*
    *SMB*
    *SMC*
    *Cinch*

This isn't ideal, for several reasons.  The `*SM?*` filters match a lot of things they shouldn't: diodes, and a few inductors.  The U.FL footprints in `Connector_Coaxial.pretty/` aren't matched at all.  Clearly something needs to be done.  We have a few options:

1. Add more types of connector to the filter lists, e.g. `*U.FL*`
2. Replace the existing filters with a library-based filter, e.g. `Connector_Coaxial:*`
3. Rename our coaxial connector footprints to have a common prefix, and replace the existing filters with a new filter matching that, e.g. `Coaxial*`

Option 1 involves the least work right now, but will require additional work every time a new kind of coaxial connector footprint is added, so I don't love it.  If we go with this, it might be best to at least remove the leading `*` to avoid matching diodes and inductors.  We'd still get trouble from e.g. N connectors.

Option 2 isn't something we've done anywhere else as far as I know, but it would work nicely for our own library.  The problem is that it would exclude footprints in third-party libraries from matching the filter of our generic coaxial connector symbol.  Really, that seems like a general problem with library-based filtering: how could we ever use it without making it hard to use third-party libraries?

Option 3 would be similar to what we've done in a lot of the other connector libraries, would make filtering easy, and would work with third-party libraries.  Overall I'd say it seems best then, except that coaxial connectors as a whole aren't a series, so it might be an abuse of our naming convention in [F3.6].  The other problem is that we'd have to rename all our coaxial connector footprints, but that isn't very many.

If we want the benefits of option 3 without renaming our coaxial connectors, we could compromise by combining options 2 and 3.  This would allow third-party libraries to have coaxial connector footprints with names prefixed with `Coaxial`, while allowing us to keep our names the same, since they're matched by the library filter.

@poeschlr @jkriege2 @evanshultz @Misca1234 Thoughts?

evanshultz, 2018-06-05:
Good find!

What about a modification of option 2 to just `Conn*:*`. Agreed it isn't something we've done, but it limits the scope nicely and isn't so specific that 3rd-party libs couldn't easily plug in. If we think 3rd-party libs would be company-specific, like `ACME_Connector` then we could try `*Conn*:*`.

With option 1, does this happen often? Do new standards arrive regularly or do we have new footprints popping up frequently? If not, this might be a one-time thing now that only needs to get touched perhaps every few years. Maybe that's not so bad if we don't want to limit the library in any way.

Ratfink, 2018-06-05:
I've seen plenty of projects with a single project-specific library for all parts that the project needs that aren't in the KiCad library.  It would be a shame to leave that configuration behind.

Plus, the filter `*Conn*:*` would match a lot of non-coaxial connectors.  So then we're stuck either placing further restrictions on library name, e.g. `*Conn*Coax*:*`, or having a list of filters like we do now, e.g. `*Conn*:BNC*`.  The former exacerbates the problem I brought up in the previous paragraph, and the latter offers little advantage over option 1.

I'm not aware of new families of coaxial connectors being created often, but it does happen sometimes, and our current list is very much incomplete.  If we go the lazy route of adding new filters only when we get new footprints that need them, we risk having third-party footprints not match our filters, and we librarians might forget to add filters when we need to.  If we try to compile make a big list of all the families we can find right now, we would still occasionally have to amend it in the future.  Plus, just imagine the fun of having a filter `C*` for [C connectors](https://en.wikipedia.org/wiki/C_connector).

evanshultz, 2018-06-05:
If the user is taking advantage of the pin count filter, are there really _that_ many footprints that will be available? I haven't done this, just wondering.

Ratfink, 2018-06-05:
A filter of `C*` on a two-pin symbol?  Gee, that sounds awfully familiar, but I don't have the *capacity* to recall…😁

While working on https://github.com/KiCad/kicad-footprints/pull/607, I just noticed that our screw terminal symbols use a library-based filter of `TerminalBlock*:*`.  Maybe a variant of option 2 such as `*Conn*Coax*:*` actually would be the best thing to go with here, then.