KiCad/kicad-symbols #652: Foot print filter for SON

Misca1234, 2018-06-04:
How is this foot print filter working.

I was uberhauling the Memory_NVRAM library and try to fix the MB85RS16 by creating the two missing foot prints

Package_SO.pretty/SOP-8_3.9x5.05mm_P1.27mm.kicad_mod
Package_SON.pretty/SON-8-1EP_3.0x2.0mm_P0.5mm_EP1.4x1.6mm.kicad_mod

(The SON is very similar to DFN-8)

But regardless how I write the foot print filter Cvpcb wont catch the SON-8-1EP_3.0x2.0mm_P0.5mm_EP1.4x1.6mm even if I set the foot print filter to "SON*" 
even then it only catch Package_SON:X2SON-8_1.4x1mm_P0.35mm

It tried first with the filter
SON*1EP?3.0x2.0mm?P0.5mm?EP1.4x1.6mm*

Ratfink, 2018-06-04:
If the footprint filter `SON*` matches `Package_SON:X2SON-8_1.4x1mm_P0.35mm`, you are using an old version of KiCad that implicitly adds an extra `*` to the beginning and end of all footprint filters.  You should update, then test this again.

Beyond that, I'd suggest carefully checking for typos in both the filter and footprint name.

Misca1234, 2018-06-05:
I found it, the one who made the original symbol was only making 8 pin symbol, the SON do have a pad on the belly side and EP1 pin is "missing" in the symbol, also, all different types with different foot prints of the the Fujitis NVRAM's MB85RSXXX is aliases of MB85RS16, it need a clean up

closed 2018-06-05 by Misca1234