KiCad/kicad-symbols #670: Make foot print filter empty for symbols with dedicated fotprints

Misca1234, 2018-06-16:
If a symbol points to a specific foot print then the foot print filter have no meaning for the symbol and should be ignored, and therefore the fp should be empty.

poeschlr, 2018-06-16:
The footprint filter in such cases is there to allow for alternative footprints.
Example a hand solder variant. Or the thermal vias variant. Or ...

Misca1234, 2018-06-16:
That is what I argue against .
If a symbol points out a specific footprint, then it is that one that should be used, nothing else.
Why would you want to get alternatives.

If you want the hand-soldering alternative, then make the "specific foot print filter" field empty and rely on the FP only.

The Cvpcb also takes a very long time when it have to deal with the FP as well, if you have 20 symbols in the schematic with dedicated foot prints then Cvpcb should be finished in less than a second.

The hand soldering argument is pretty weak, now when all components get smaller and smaller, 
that argument was strong back in the early 80'th when DIL was the only choice of package.

poeschlr, 2018-06-16:
People are not forced to use an alternative footprints but who are we to disallow for that? (It does not hurt the purists but it allows for more workflows to use the same symbols.)

And in kicad 5 the footprint filter is not only useful in the cvpcb workflow but the new experimental component chooser uses them as well.

---

I regularly handsolder smd pcbs. Down to 0402. (and sometimes even qfn but i try to avoid that and mostly choose qfps instead.) Handsoldering footprints simply have their place. Just because you do not use it does not mean others should not be allowed to do it.

Misca1234, 2018-06-16:
Let me reformulate the core issue for the thread.

If someone submit a symbol and have specified a specific foot print, then it should not be mandatory to fill out the foot print filter as well.

> in kicad 5 the footprint filter is not only useful in the cvpcb workflow
That is quit extraordinary, if a component have a specific footprint then it sounds a little magic that cvpcb can do something about it, or even try to do something else, which it should not do in the first place.
If the component does not have a specific chosen footprint but only a FP filter then then it sounds more reasonable.

> but the new experimental component chooser uses them as well.
If a component have a specific chosen footprint, why would the component chooser mess around with alternative foot prints?
If a component only have a FP then the component chooser can show alternatives

An alternative way is totally remove the possibility to specify a specific foot print and only rely on the FP.

evanshultz, 2018-06-30:
I am with Rene on this. The footprint filter has obvious uses even with a default footprint. Only Jan hasn't weighed in but that's still the majority of us. It's always good to ask questions but I think this issue can be closed.