KiCad/kicad-symbols #692: RC3 will be tagged on sunday early afternoon (central european time)

poeschlr, 2018-06-24:
Wayne announced a few hours ago that he wants to tag rc3. Meaning we will need to tag the lib as well.

The idea is to publish v5 on the 30th. As i am in London (probably without access to the internet) for the first week of july, i can really only tag the lib if the release falls onto the 30th. Otherwise one of you guys would need to do it.

evanshultz, 2018-06-24:
Thanks for the heads-up Rene.

I'd appreciate if somebody could review https://github.com/KiCad/kicad-symbols/pull/641 before then. It adds a new lib which, even if it's not in the packaged library, will still be in the library available from KiCad's website by the release date.

poeschlr, 2018-06-24:
Could somebody look at my exposed pad size fixes sometime this week. Would be really good if this could make it into v5.

evanshultz, 2018-06-25:
Sure. I expect to get to them Tuesday if nobody beats me to it.

poeschlr, 2018-06-26:
https://github.com/KiCad/kicad-footprints/pull/689 and https://github.com/KiCad/kicad-footprints/pull/690 should be easy decisions as they are basically already reviewed by @jcyrax (They would also be major improvements. Especially for tantal smd footprints.)

evanshultz, 2018-06-26:
I'll look at those first. There are quite a few you've submitted on this topic, though. Do you have a priority for any others?

evanshultz, 2018-06-28:
The important two mentioned above are done. I'll start on QFN stuff tomorrow. I would like to do https://github.com/KiCad/kicad-symbols/pull/661 next. Do you have any priority?

evanshultz, 2018-07-01:
RC3 is now tagged: https://lists.launchpad.net/kicad-developers/msg36380.html. Only one week late! :) Anything else high-priority that needs to get done this next week?

evanshultz, 2018-07-03:
@jkriege2 @Ratfink @Misca1234 
If anybody is available, here are a few PRs I'd like to see get done before 5.0 release (10 Jul).

Sensor_Voltage footprint lib added:
- https://github.com/KiCad/kicad-footprints/pull/714
- https://github.com/KiCad/kicad-symbols/pull/717

Significant MSP430 lib cleanup and missing footprints added:
- https://github.com/KiCad/kicad-footprints/pull/718
- https://github.com/KiCad/kicad-symbols/pull/710

pointhi, 2018-07-05:
I like to drop one of my PR also into this list: https://github.com/KiCad/kicad-footprints/pull/716

I think it is important, because the current footprints can cause unconnected warnings

Ratfink, 2018-07-05:
A minor thing I'd like to see before the release: renaming `SOT-89-5_Housing` to `SOT-89-5`.

- [ ] https://github.com/KiCad/kicad-symbols/pull/732
- [ ] https://github.com/KiCad/kicad-footprints/pull/731

Shackmeister, 2018-07-05:
I still have a few renaming issues in the 3D models. Problems is, it's typically me and @poeschlr who takes care

Shackmeister, 2018-07-05:
I still have a few renaming issues in the 3D models. Problems is, it's typically me and @poeschlr who takes care of the 3D PR's and since he is in London I could need Someone to merge my work. Perhaps @easyw could do the review and then somebody could press the merge button?

Ratfink, 2018-07-06:
More #520 fixes that I'd like to get done in time for the new stable release:

- [ ] https://github.com/KiCad/kicad-symbols/pull/735
- [ ] https://github.com/KiCad/kicad-footprints/pull/732

poeschlr, 2018-07-12:
@Shackmeister  i can not see any open pull request on the package3d repo that is by you. Have they already been merged?

Misca1234, 2018-07-12:
Yes, I merged them

evanshultz, 2018-07-14:
Any chance to get one or both of the following merged before the libs are tagged for 5.0.0?
- [ ] https://github.com/KiCad/kicad-symbols/pull/747
- [ ] https://github.com/KiCad/kicad-symbols/pull/754

We can close this PR regardless, I think, because RC3 is over and 5.0.0 source is tagged.

poeschlr, 2018-07-15:
I would like to get https://github.com/KiCad/kicad-footprints/pull/746 into v5 if possible.

I think as we already list stuff for between rc3 and final v5 in this issue we can keep it open and manage v5 using this issue.

evanshultz, 2018-07-16:
@poeschlr 
Done!

thomasonw, 2018-07-17:
I would offer issue #756 ([Hidden power pins](https://github.com/KiCad/kicad-symbols/issues/756))   should be addressed before going to Ver 5 release.

poeschlr, 2018-07-17:
Please only post pull requests here. We will not postpone the v5 release for any of the open issues.

poeschlr, 2018-07-17:
I would like to tag v5 sometime on wednesday eveneng (CET) such that the packagers have a bit of time till the v5 release announcement. It seems 5.0.1 might be released very shortly after the v5.0.0 release so we can have some bugs that are fixed at that point in time.

@evanshultz as i stated in my comment of the opamp fixes, we can merge them as they now stand and leave the rest for v5.0.1 if you run out of time.

---

On a different note. These would also be quite nice to be in v5
- [ ] https://github.com/KiCad/kicad-footprints/pull/753
  - [ ] https://github.com/KiCad/kicad-symbols/pull/764
  - [ ] https://github.com/KiCad/kicad-symbols/pull/763
- [ ] https://github.com/KiCad/kicad-symbols/pull/762
  - [ ] https://github.com/KiCad/kicad-footprints/pull/752

And my general effort to transfer qfn footprints over to be scripted:
- [ ] https://github.com/KiCad/kicad-footprints/pull/751

evanshultz, 2018-07-17:
Because GH isn't happy with the branch, re-submitted with fixes at https://github.com/KiCad/kicad-symbols/pull/767. Sorry the history is still messy; will fix soon.

I'll also look at the PRs you mentioned above in the next 12 hours.

poeschlr, 2018-07-19:
I will create the tags for v5 today evening (at the latest Thursday 23:59 CEST)
If any of you want something to be reviewed by then please post it here.
I will aim to be available for kicad work from 19:00 CEST till i create the tags.

evanshultz, 2018-07-19:
@poeschlr 
All of the QFN PRs above are merged.

poeschlr, 2018-07-20:
The libs are now tagged for the version 5 release. Onward to KiCad 5.0.1 or 5.1.0 (depending on how fast the wxwidget stuff can be updated to gtk3 based)

I would still keep the lib as stable as possible. (So no reorganizations or renamings of libs. Footprint and symbol renamings only to fix bugs but not to change the naming convention.)

closed 2018-07-20 by poeschlr