KiCad/kicad-symbols #698: No generic dual diode

evanshultz, 2018-06-25:
We have a few dual diodes in Diode.lib (BAS40, BAT54C, MMBZxx) but there isn't a generic in Device.lib. I can't recall other diode symbols with a generic version so I suppose this is an oversight.

ioch, 2018-06-27:
There are couple D_x2_* in Device.lib. The name is a bit cryptic and hard to find but they are there

evanshultz, 2018-06-30:
@ioch 
Oh yes. There they are. Thank you.

@poeschlr @jkriege2 @Ratfink @Misca1234 
Is `x2` obvious to you? Not to me. I think `Dual`, `Triple`, `Quad`, etc. is much easier to grok.

Over at https://github.com/KiCad/kicad-symbols/issues/580 we talked about uses for both versions. And for resistor arrays, using words going up to 8 (through `Octal`) gets a bit weird and `x8` seems better to me.

What do you all think? Is there some rule we can formulate? Perhaps something like:
```
Prefer words (`Dual`, `Triple`, `Quad`) if the number of units is 4 or less.
For a greater number of units, use `x<number>` format.
If both are needed in the same part, such as a quad switch where two actuators each operate a pair of switches, use words first and then numbers.
```

It sounds awkward but that's the best I have now. Is there a better wording or better method?

Ratfink, 2018-06-30:
I see no problem with the current naming of dual diode symbols.  "Dual" is in the description for all of them, so searching for "dual diode" shows all of the symbols you'd expect.

I do, however, suggest we rename the `LED_Dual_*` symbols to match the other dual diode symbols' naming, preferably within the week so the 5.0.0 release can have consistent generic diode symbol naming.