KiCad/kicad-symbols #719: Splitting microchip SAM microcontrollers into multiple libraries is not optimal

HubertD, 2018-07-01:
Currently, there are libs for Microchip SAML and SAME devices; 
I have pull requests open for the SAMC and SAMD families.

Currently working on more SAMD devices, I found out that some of them share their pinouts with SAMC devices. Adding them to the SAMD library would mean to copy the symbol from the SAMC lib.

If we want to avoid copying identical symbols, we would have to join SAMC and SAMD libs (and probably more of them; SAML devices also seem to be compatible, SAME don't).

On the other hand, joining all of them might create a very large library in the end (not sure, maybe more than the 250 items specified in KLC G1.2 if that number includes aliases).

How shall I continue?

evanshultz, 2018-07-03:
It's a good question. Thanks for bringing it up. Why make the library file sizes larger when ALIASes within a single library could be used, right?

Within the embedded MCU section, in just T&R packaging, Digikey carries the following parts in each family:
- 100 SAMC
- 226 SAMD
- 51 SAME
- 115 SAML

I think that's enough for separate libraries already. Plus there's SAMG, SAMS, SAMV, SAM3, SAM4, SAM7...

With the push towards atomic symbols, making more unique symbols will become more and more common throughout. And the file sizes of libraries are rather small anyway.

Plus, using script generation of symbols means less manual checking and easier updating so I hope to see that more and more. Scripts mean manual overhead of manipulating individual symbols goes away. Tweak a script, squirt out new files, and update the library. Easy.

I generally would like to join symbols together and have smaller file sizes and fewer libraries, but since there are so many SAM families and they've already been split (not just a single SAM library), I vote for continuing along this path.

HubertD, 2018-07-03:
okay by me :-)