KiCad/kicad-symbols #723: Library for generic driver symbols

evanshultz, 2018-07-03:
See https://github.com/KiCad/kicad-symbols/pull/639.

This symbol is definitely a driver, but there's no specific part to be driven. I think this part would be best in a Driver_ library. Maybe Driver_Motor or Driver_Relay? This part is similar to DRV8860 which is in the latter one.

That being said, just a generic Driver library may be best since some of these are general purpose drivers; I don't know why there isn't a generic one. Would it be best to have a generic `Driver.lib` for this kind of part that isn't specifically for driving motors or FETs or LEDs or...

myfreescalewebpage, 2019-01-26:
@evanshultz agreed with you with `Driver.lib`.

antoniovazquezblanco, 2019-06-27:
The mentioned symbol was merged in `Transistor_Array`.

Is the discussion about creating a `Driver.lib` open?

evanshultz, 2019-06-27:
I still think a generic `Driver.lib` would be useful.  With a review of existing drivers perhaps there's enough support to populate this new library. Let me take a quick look and see how many might be candidates for a generic driver library...

evanshultz, 2019-07-15:
- Driver_Display:
-- This is a pretty specific category; a quick check of the symbols shows that this is fine as-is.

- Driver_FET:
-- Seems like these are all valid and the library is substantial and a reasonable subdivision of driver ICs.

- Driver_Haptic:
-- The single symbol here is really a motor driver, but intended for haptic applications. So it's mis-classified and just a single symbol isn't enough to warrant a whole library, IMO.

- Driver_LED:
-- I didn't look at every single symbol, but there are enough LED-focused symbols to be worth their own library. This is a specific use case with ICs finely-tuned for this just purpose.

- Driver_Motor:
-- A worthwhile category for specific ICs with pins/features to control direction, step, etc. of a motor.
-- L293 and L298 are generic (L297 has motor-specific features).

- Driver_Relay:
-- All three symbols this in like can be used for many applications.

---

So IMO, Driver_Haptic and Driver_Relay, along with some Driver_Motor symbols, would fit best in a generic `Driver.lib`. Without knowing if there's support for this, I didn't do a symbol-by-symbol review. Is there agreement to make a `Driver.lib` the next time it could be used, and then I can do more digging to figure out which existing symbols should move to `Driver.lib` at the next opportunity?

Also note that many of the symbols in these libraries are old and in need or major work to make them meet KLC and visual align with symbols current being merged.