KiCad/kicad-symbols #743: Issues with jumpers

evanshultz, 2018-07-09:
This is only intended to spark conversation. Some of this is thinking out loud so please read and then add comments. No need to offend and I'm not strongly married or opposed to any ideas.

https://github.com/KiCad/kicad-footprints/pull/716 made me look at and think about jumper symbols which would match these footprints.

First, there are jumper symbols in two libs: https://github.com/KiCad/kicad-symbols/blob/master/Device.lib and https://github.com/KiCad/kicad-symbols/blob/master/Jumper.lib. Shouldn't there just be one location for them?

They look like this:
[image]

One of the core tenants of KiCad is having generic symbols with a footprint selected after placement using CvPcb. There are some atomic symbols, but jumpers aren't one of them.

So that leads me to several points:
1. Jumper.lib has solder bridge and non-solder bridge symbols. Why do we care? Why isn't the symbol as generic as possible with all filtering of physical part done in CvPcb?
1. If the above is taken to the extreme, having bridged symbols doesn't make sense. There could just be a 2-pin and a 3-pin jumper symbol, with pin numbers shown, and then CvPcb shows all shorted and bridged footprints with the pin numbers allowing matching with semi-bridged footprint choices.
1. Symbols that are not shorted still have a small arcs between the pins. Why? This does not guide footprint selection or the logical connection in any way. Only a design-specific note on the schematic could indicate if an "open" jumper would actually be shorted in some specific schematic.
1. The Jumper.lib symbols are excessively large to avoid 50mil pins. There's no reason for this as the function is clear even if the pins aren't long.
1. Symbol naming is inconsistent between the libraries where jumper symbols are found.
1. The Device.lin symbols have no footprint filters and several of the Jumper.lib ones don't match any footprints either (`SolderJumper_3_Bridged123`, `Jumper_3_Bridged12`, `Jumper_2_Open`, etc.)
1. We could be more concise about terms. Instead of "Bridged" we could use "Short".
1. Shouldn't the shorted footprints be "illegal"? They will create intentional net shorts on the PCBs which aren't captured in the netlist. The IPC-D-356 netlist can show net shorts but it doesn't list these. (I can file a bug with the devs.) If board short tests are done they will measure a short that isn't on the netlist which means communicating this must be handled outside any automated method inside KiCad or else the bare board won't match the netlist connectivity!

Maybe there are more things to consider? That's just what I could quickly rattle off. And all but the last one seems relatively easy to correct.

So, all of this considered, here are my thoughts:
1. All jumper symbols are in Device.lib as they're a generic symbol.
1. Jumper symbols may use 50mil pin lengths and other deviations we allow for small symbols.
1. Pin numbers are shown on all jumper symbols.
1. Short or open configuration is not indicated by the symbol.
1. The physical implementation of the jumper (solder, wire, etc.) is not indicated in the symbol name.
1. Symbol name format is simply `Jumper-<pin count>`.
1. Footprint filter is only `Jumper*` as the physical jumper implementation doesn't matter to the symbol. Existing jumper footprints will need to be renamed.

This will result in just 2 symbols which each only have a handful of footprints when using the keyword and pin count filter in CvPcb. This is quite manageable and much easier than the current situation.

Unfortunately, since there is already a dedicated symbol library for jumpers perhaps it's too late to go back on the first item?

jkriege2, 2018-07-09:
Hi!

since I made these symbols a long while ago, some comments from me:

1. I think the jumpers should stay in Jumper.lib ... as we also have swicthes.lib, which also contains generic symbols.
2. IMHO: Size should be the same as the existing switch symbols (to allow for easy replacement)
3. pin numbers would only be necessary for jumpers with >2 pins (makes documentation easier, if it's clear in the schematic what 1-2 or 2-3 mean!
4. I would object, especially for the SolderBridges, where there are variants that actually have copper between the pads to make a default connection (I agree that we would need a software tweak to encode that internal connection to allow for testing!) ... for "normal" jumpers: often you have default settings and indicating those in the schematic, can make it much more readable (not in all cases of yourse)!
5. Hhmmm ... again, for documentation purposes I think symbolizing at least those two basic types in teh schematic may help ... think e.g. of a board where there are several of both types. it can make design/documentation much easier in that case
6. see 5
7. that only makes sense if the differently bridged version are not there!

Best
JAN

evanshultz, 2018-07-09:
2. You mean physical switches? Like a 1P1T toggle switch, for example?
4. Is this personal preference? Why would a "solder bridge" symbol be different than any other jumper symbol which could have pins tied together with copper too? If solder is making the connection, there is no physical part associated with the jumper so the part need not appear in a BOM as a physical part (I'm used to work instructions being separate and if the solder is applied during wave it's "automatic" for manufacturing). Forcing unique symbols for open and short seems antithetical to the KiCad workflow at http://docs.kicad-pcb.org/stable/en/getting_started_in_kicad.html#kicad-work-flow by having generic symbols which are overly specific. Maybe I'm wrong?
5. What is the other basic type? A bent metal wire jumper going into THTs?

jkriege2, 2018-07-09:
- ad 2: yes
- ad 3 (I think you answered on 4?): I'm thinking about "bridged/closed solder bridges" (for lack of a better name) as they can be found in the footprints, where two contacts are connected by a thin strip of copper that can be cut with a knife (subsequently the other contact can optionally be closed by solder, if it's present). ... and YES, there does not need to be a physical part for these in the BOM, but it's handy to have them as footprint, when designing the PCB. THe basic type for me would be a header with a jumper put on it.

Best,
JAN

LarryJoy, 2018-09-04:
Okay, let's run through this. I have seen the class letters "JP" used and presumably this means jumper, however, this is not in the list of "standard" class letters of IEEE 315, Clause 22.4. How should "jumpers" be handled?
1) For the case of having a header with male terminals this connector would use class letter J. The plug with female terminals, used as the "jumper", would use the class letter P. If you have a multi-terminal header (more than 3 terminals) that would use 2 or more plugs, each jumper plug would be assigned a ref des of P#. Remember that P numbers do not have to match with J numbers. (See ANSI/ASME Y14.44, Clause **2.1.5.3 Connectors** and **Fig. 5 Typical Connector and Interunit Cable Designation**.) The "jumper" plugs will not have a land pattern/footprint.
2) For the case of a piece of wire used as a "jumper" that is inserted into a pair of THT plated thru holes or soldered to SMT pads, the class letter W applies.
3) For a solder jumper I have seen the class letter(s) of SJ used. The graphic symbol to use is the terminal symbol, which is a non-filled circle. The ref des SJ#s would not appear on a parts list (PL) as they are part of the PCB and not separate items.

sslupsky, 2018-10-10:
One issue to consider regarding bridges.  I am using one of them.  However, the bridge fails DRC in pcbnew.  I though of using an open bridge and then using the ECO2 layer as my "bridge layer" to make the default bridge.  However, I know of no way to combine the ECO2 layer with the F.Cu layer when creating the gerber.  Can the DRC rule be suppressed for the bridged jumpers?

poeschlr, 2020-01-27:
I would say this is a topic for version 6 (so i set the milestone for it). I would vote for having jumpers only in the jumper lib and possibly also adding symbols for jumpers using the typical pin headers (something that looks like such a jumper side on.)

sslupsky, 2020-01-27:
Please consider a DRC rule for these as well.

poeschlr, 2020-01-28:
V6 will have native support for net ties. This will solve such problems. Plus our current version of net ties does not create a DRC error so you might want to look at them to learn how to solve your problems.

cpresser, 2020-09-17:
I am currently looking at all things tagged v6 and this came up.
At least part of it seems solved: I cant find jumpers in `Device.lib`.

I would rework the pin positions. At least the symbols in `Jumper.lib` should be interchangeable. Currently they are not:
[image]