KiCad/kicad-symbols #749: Symbol naming for variations of generic symbols

evanshultz, 2018-07-09:
When making https://github.com/KiCad/kicad-footprints/pull/738, I find that none of the existing dual LED symbols have a matching pinout for K[ingbright AAA3528ESGCT](http://www.kingbrightusa.com/images/catalog/SPEC/AAA3528ESGCT.pdf):
[image]

So I can add more generic symbols, but I'm not sure that's super productive either.

https://github.com/KiCad/kicad-footprints/pull/597 is another example where we have a generic symbol but that symbol doesn't work with a specific component's footprint.

What do we do in these cases? For dual LEDs and common mode filters, we don't have all permutations. For bridges diodes and SCRs, for example, we have every variation I can think of.

Sometimes the symbol name is helpful to understand what actual component could use the generic symbol but other times the symbol name isn't too helpful. Having `<some symbol name>` and `<some symbol name>_ALT`, for example, isn't descriptive. Many of them have the order of the pins as the differentiating element of the generic symbol name (LEDs, Q_DUAL*, etc.) and I suppose that's not bad if we want to prioritize brevity in some form.

So all in all, I don't feel like the many permutations of generic symbols are named helpfully. Does anyone else feel the same way? I don't have any great ideas right now, but if there is a consensus to try and make symbol names when we have a myriad of generic symbols more clear then we can think about it.

karlp, 2018-11-08:
for generic things, can they just not have pin numbers? can that just be something for the footprint assignment to take care of?

evanshultz, 2018-11-08:
In KiCad at the current time (5.0.x release), pins on symbols need some type of unique designator (name or number). While the pins of a resistor are interchangeable (usually), consider a diode. The anode and cathode will need to be known in the schematic so that when a footprint is selected the cathode pin of the symbol goes to the cathode pad of the footprint.

Please give more info if you had a different idea.

fauxpark, 2018-11-08:
Is this at all related to #861?

karlp, 2018-11-08:
I was just saying that names like Anode1 and Anode2 are perahps more useful than explicit pin numbers, as the symbolic names can be assigned to any numbering scheme in the footprint assignment, that's all

myfreescalewebpage, 2019-01-15:
Hi @evanshultz, symbol has been proposed since you have written this issue: https://github.com/KiCad/kicad-symbols/pull/1068

Proposition to close this issue once it will be merged.

Cheers,
Joel

myfreescalewebpage, 2019-03-23:
Closing following merge of #1068

closed 2019-03-23 by myfreescalewebpage

evanshultz, 2019-03-23:
My proposal is to change from pin numbers, which result in a lot of various symbols, to pin names, which would require only one symbol. Then we have various footprints which use the pin names and match. Ultimately, there will then only need to be a single symbol so it's much easier on the user, I believe.

In this case, the pin "numbers" could be A1, K1, A2, and K2. We had a discussion and this was accepted for cards (SD, SIM, etc.) for this reason.

I'm suggesting this should become more widespread in the library, perhaps a target for v6, so that fewer generic symbols are needed throughout the library.

myfreescalewebpage, 2019-03-24:
Clearly not bad idea @evanshultz this can be applied for transistors, potentiometers, leds, transformer etc etc :)