KiCad/kicad-symbols #786: Atmega328 is missing pin 22

Mek7, 2018-07-25:
Atmega328 is missing pin 22 (GND). I don't know if it was left out intentionally since there is already one GND, pin 8. But I would like to connect these two to GND in schematic to make it clear and so that it makes the correct ratsnest for PCB design.

poeschlr, 2018-07-25:
Which one? There are several Atmega328 symbols.

Mek7, 2018-07-25:
Pin32? Did you mean pin 22?
Yes, that is true, now I see that when I got to designing the PCB, ratsnest is correct. Anyway, it's confusing...

poeschlr, 2018-07-25:
It is just solved this way because we can not assign multiple pads per symbol pin. Maybe the new format will give us that feature but it is at least two years away. 
There are always heated discussions about this particular topic. The two sides of the argument are well known and boil down to user preference and what tools one used before kicad.

closed 2018-07-25 by poeschlr