KiCad/kicad-symbols #805: Inadequate SPICE symbols

evanshultz, 2018-08-08:
There are a few issues I see with this library:
- Only a smattering of symbols are present
- Many symbols are very unlike the standard symbols in other libraries
- The size/shape/style of some symbols stands out (in a bad way)
- It would be nice to have some pre-configured voltage/current source symbols.

I will work on this.

@poeschlr 
This will require adding relevant fields to the symbols which would seem to violate KLC (see https://github.com/KiCad/kicad-symbols/issues/464#issuecomment-411449086), but KLC was written without SPICE simulation in mind and this is a special case so I think an allowance just for the simulation library is fine. Please let me know if this is not acceptable.

poeschlr, 2018-08-08:
As long as this stays (for now) within some specialized libs i am ok with that. I have absolutely no expirience with regard to spice and i really do not intent to learn it within the next few months. (Have too much other stuff to do)

poeschlr, 2018-08-30:
Just to add to this: please keep the current symbols alive in a way such that updating the lib would not break exisiting projects. (meaning there should be symbols with the same name and pin positions/pin numbers in the lib after your change)

evanshultz, 2018-08-30:
Yes, I understand. Honestly, I don't know who made these symbols or what they were thinking. They're so unlike the rest of the library, do not cover all SPICE primitives that the ngspice engine supports, and most are just plain ugly. And I already brought up my concern with the library name.

poeschlr, 2018-08-30:
From the git history of this lib in the old repo i gather that this lib existed since the beginning of the library here on github. Most likely it existed long before that. My guess would be it was used to test some early spice interface and never got updated as nobody needed it. Now that kicad supports simulation people will start to use it and notice how bad that lib is.

poeschlr, 2018-08-31:
Just to keep all information centralized here a repeat of my answer over at the mailing list:

Instead of simply renaming the existing pspice lib i would suggest to declare it deprecated. (Remove it from the default sym-lib-table). We might also want to add a "deprecated libraries" file that tells travis (and users) that some library is left out of the lib table on purpose.

I would then create a new lib with symbols that use all kicad 5 simulation specific features. It might be possible to make these symbols in such a way that they do not contradict our footpritns (pin number assignment for diodes for example. Here the spice pin number remap feature should be used instead of assigning the pins in the wrong order for footprints)

pgrudzinski, 2018-10-28:
I would be glad to help with that. Is there anything done on this subject right now? I do not want to duplicate something that is ready/being worked on.

evanshultz, 2018-10-28:
Thanks for the offer. I've already started on the sources, which I think is the bulk of the missing symbols.

evanshultz, 2019-02-08:
Sources will be included in a new Simulation_SPICE lib in 5.1. See https://github.com/KiCad/kicad-symbols/pull/1473.

Next up:
- Switches
- Transmission Lines

Then Linear Dependent Sources.

And finally will be Non-Linear Dependent Sources.

Coupling (K) is an important one, but I think that's best left as text for the user to place on the schematic (like .mea, .tran, .include, etc.) with the current implementation of circuit simulation in KiCad.