KiCad/kicad-symbols #808: Reference designator

leoheck, 2018-08-09:
According to the [Kicad Library Conventions](http://kicad-pcb.org/libraries/klc/S6.1/), the reference designator for Ferrite Beads is `FB`. Right now, Kicad Library has it as `L`, which is the same as inductors. Which one is the right one?

evanshultz, 2018-08-09:
Good question. Probably an oversight.

KLC mostly takes the ref des prefix from IEEE 315, but ferrite bead isn't listed there ("ferrite bead ring" is, but with a ref des prefix of `E`).

In light of any official standard, I prefer `FB` and updating the symbols in the library.

leoheck, 2018-08-09:
Alright! That is good, I am using `FB` too. I agree that it to the main library is better than keeping the `L`

LarryJoy, 2018-08-28:
[image]
What is the difference between a "ferrite bead" and a "ferrite bead ring"? FB is not a class letter in IEEE 315, Clause 22.4 list. As stated above about class letter E, that is what I use. And the "ferrite bead" symbol is from IEC 60617/IEEE 315A, Clause 6.2.11 "Ferrite bead, shown on a conductor".

poeschlr, 2018-08-28:
We are not the only people discussing the ferrite symbol/ reference designator:
https://electronics.stackexchange.com/questions/173656/north-american-standard-symbol-for-ferrite-bead
https://www.pcblibraries.com/Forum/ferrite-bead-symbol-and-reference-designator_topic2271.html

LarryJoy, 2018-08-28:
Further to the discussion of the class letter E. There are 32 items the class letter E stands for:
[image]
Further, the heading of IEEE 315, Clause 22.4 has:
[image]
If a ferrite bead does not fit a ferrite bead ring then still use class letter E to mean miscellaneous electrical part. What class letter would you use for a ferrite binocular core? I would use E, why use two letters when one will do?

poeschlr, 2018-08-28:
IEEE has as far as i know the limitation of using only single letter refdes prefixes. (This limits them to at most 26 different groups of symbols. So there sill be a lot of overlap)
Also remember that standard is very old already (Not sure if i got updated but the first release was in 1975.)

We are not really bound by that restriction. FB seems to be widely enough used in the industry that it appears in Wikipedia (and many other online sources) so it should be ok for kicad.

LarryJoy, 2018-08-28:
A basic reference designation (ref des) consists of a class designation letter (or just class letter) and a number (a serial number) such as R1. For a complete reference designation such as A3A1A2R1, the A3A1A2 are referred to as reference designator (ref des) prefixes. A class letter can be 1, 2, or 3 letters but if 3 letters the first letter will be an X, such as XAR (a socket for an op amp), XDS (a socket for an LED or display), or XBT (a battery holder). IEEE Std 315 is no longer an ANSI standard but the IEEE still has it as active, but do not maintain or support it, as you can still purchase this standard from them. ANSI/ASME Y14.44 (that replaced IEEE Std 200) "Reference Designations for Electrical and Electronics Parts and Equipment" calls out IEEE 315, Clause 22 for class letters, so in this way it is still the "standard". IEEE Std 315, or whatever predecessor, came out before 1975.

The reason people use oddball class letters is because they are unaware and were not taught properly that there was/is a standard. I tell the story of the traffic sign that has an octagonal shape. It doesn't matter if the sign has ALTO, HALT, or STOP on it, everybody knows what it means. If you were driving around on a dark night and saw a rectangular sign with STOP on it, it would make you ponder. Remember the saying "Don't believe anything you hear and only half of what you read". Just because it is on Wikipedia doesn't make it so. Where did they get their information--be specific, name your reference.

The IEC has 81346-1 and 81346-2 that uses two letters for everything in functional designations. I just saw an announcement that the next revision is ready and will be released by the end of 2018. They are going to three letters for everything. I don't think you want to go there.

poeschlr, 2018-08-28:
> The IEC has 81346-1 and 81346-2 that uses two letters for everything in functional designations. I just saw an announcement that the next revision is ready and will be released by the end of 2018. They are going to three letters for everything. I don't think you want to go there.

Why would you think that we do not want to go there? 

---

If a standard is not really followed that strictly in the industry why should we follow it strictly. Our purpose is not to teach users the proper interpretation of some aging standard but to provide a useful library that is recognicable by somebody working in the field.

To me it seems the industry uses a mixture of different standards. They use single letter prefixes for common parts that had a good clear assignment in the old standard but also include multi letter prefixes from newer standards where the old standard did not provide a satisfying option. (Ferrite beads are one of these things without a good option in the old standard)

From what i can tell, FB is a very common reference designator prefix for such ferrite beads (common enough that google finds more results for it then for E)

evanshultz, 2018-08-28:
"Ferrite bead ring" sounded like either the clamp-on style that go around a cable or the little cylinder that goes around a lead (like an Amobead on a TO220) to me. To get those on a BOM it may need a ref des. And `E` included a number of things that aren't necessarily PCB-mount electrical components.

Anyway, FB it is. PR submitted at https://github.com/KiCad/kicad-symbols/pull/854.

closed 2018-08-28 by evanshultz

LarryJoy, 2018-08-28:
poeschlr,
The IEC calls their method a Reference Designation System (RDS). Using the method you can get to a city, a street in the city, an address on the street, a building at the address, a floor level in the building, a room on the floor, a cabinet in the room, etc. It is very extensive.

They use, at least right now before the next revision, a two-letter class code. The first letter is class and the second letter is subclass. Take for instance the class code "RA". The intended purpose or task of object R is "Restricting or stabilizing motion or a flow of energy, information or material". The definition of the subclass is "Limiting flow of electrical energy". Examples of components are "Resistance, reactance coil, and diode". That means a resistor, an inductor, and a diode would all use the same class code. I'll leave it up to you if you want to pursue the IEC method further.

poeschlr, 2018-08-28:
Or we do it like the rest of the industry and just wing it ;)