KiCad/kicad-symbols #809: Isolated Resistor Network

leoheck, 2018-08-09:
I don't have this kind of part on my lib. It will be nice to add it.

[image]

Something like this!

[image]

evanshultz, 2018-08-09:
They already exist in the HEAD rev of the library at https://github.com/KiCad/kicad-symbols/blob/master/Device.lib:
[image]

We don't take requests here so I'll leave this open for a couple days before closing it. We do, however, welcome submissions. There are plenty of symbols that could be added so if you want to submit a PR instead of a request that would be great.

Ratfink, 2018-08-09:
Evan got in before me and covered most of what I was going to say.  However, I have one more observation:

One thing we should do here is add the keyword "isolated" to those symbols, since that does seem to be something manufacturers use to refer to these ([Bourns](https://www.bourns.com/pdfs/4100R.pdf), [Vishay](http://www.vishay.com/docs/31511/31511.pdf)).  I'll submit a PR for that shortly.

leoheck, 2018-08-09:
@evanshultz thanks to show me that. @Ratfink that is a good thing too. The image I added here on my first post is from a datasheet and has the `isolated` keyword.

closed 2018-08-10 by evanshultz