KiCad/kicad-symbols #813: Bridged jumper vs net tie symbols/footprints

evanshultz, 2018-08-10:
This is about the logical use of all library parts, including symbols and footprints. Since the design workflow starts with symbols, I've put this question here.

We have net tie symbols which allow connecting two nets together.
There are also bridged jumper symbols which do the same thing.

Why do we need both? It doesn't make sense to me that there would be two symbol/footprint options to do the same thing. Should both be retained or should the official KiCad library only have one method to join nets?

Ratfink, 2018-08-11:
As I see it, a net tie is meant to *permanently* tie nets together, whereas a bridged jumper is meant to tie nets together *by default*.  The jumper may be cut and re-connected, whereas the net tie isn't meant to ever be cut.  Note also that jumpers have bare copper, but net ties are all covered with soldermask.

poeschlr, 2018-08-31:
@evanshultz is this resolved or is there some action needed?

myfreescalewebpage, 2019-03-15:
@evanshultz that's something used by some people who find it convenient but sure everybody do not make the difference on the schematic. I propose the issue should be closed. Joel

myfreescalewebpage, 2019-03-23:
@evanshultz ping? Thanks!