KiCad/kicad-symbols #842: Missing Datasheet

R-INYURU, 2018-08-23:
In case of a datasheet missing for a symbol what can one do, if he/she needs it for the design.
Probably this is not an urgent issue to fix, but your ideas are welcomed.

poeschlr, 2018-08-23:
I am not sure what the question here is. Is there a symbol in the library without a (valid) datasheet but you would need that datasheet or do you want to submit a symbol for a component where you do not have access to a datasheet?

The former is to be expected. We can not ensure that all symbols have a valid datasheet assigned at all times. Right now most parts should have at least some datasheet assigned. (This is checked by our test scripts)

For the later: you can have anything in your local libraries. For committing it here we really need some reference against which we can check your contribution. (How should we otherwise ensure the symbol or footprint is correct:?) This reference must then also be added to the symbol or footprint such that users can also verify its correctness. This is what the datasheet field should be used for.

antoniovazquezblanco, 2018-11-07:
@R-INYURU 

Could you please post here which symbol are you talking about? Also, has your question been resolved?

Thanks!

myfreescalewebpage, 2019-01-15:
@poeschlr @antoniovazquezblanco I think @R-INYURU was saying some parts have no datasheet available from KiCad interface. That's at least the case for generic symbol and also for some few others in the library, not always checked in the past.

@R-INYURU if you want to enhanced some parts of the lib by adding or fixing datasheet links, that's of course welcome by submitting a PR :)

Proposition to close this issue at the moment.

Joel

closed 2019-01-17 by myfreescalewebpage