KiCad/kicad-symbols #856: The 5.0.1 release is planned for end of septemper. Anything we want to have included in it?

poeschlr, 2018-08-28:
https://lists.launchpad.net/kicad-developers/msg37283.html

poeschlr, 2018-08-28:
@evanshultz, @Ratfink, @jkriege2, @Shackmeister 
How strict have you guys been with avoiding merging stuff that would break existing projects? If we have been strict enough then we can include an updated library into the next bugfix release. Otherwise that release will continue using the original version 5 release (Meaning no bugfixes and no new additions will make it into kicad until version 6 is released.)

My intent was to be as strict as described in this mailing list message. Could you read it and tell me if we to the best of our knowledge fulfill the promises given in it?
https://lists.launchpad.net/kicad-developers/msg37312.html

evanshultz, 2018-08-30:
While I haven't checked anything, I don't recall any major changes off the top of my head. There has been a lot added and fixed, but it's all incremental in my recollection. I believe any breaking changes should be minor.

poeschlr, 2018-08-30:
I plan on writting a script that will give us a list of symbols that have design breaking changes. This list should allow us to make an informed decision.

poeschlr, 2018-08-31:
I found we have the following design breaking changes made in the symbol lib since the 5.0.0 tag

Deleted or renamed symbols:
- 'Driver_FET.lib:IR2106'
- 'Regulator_Linear.lib:MIC5205-ADJ'

Symbols with moved, renumbered or removed pins:
- 'Driver_FET.lib:FAN7371'
- 'Driver_FET.lib:FAN7842'
- 'Driver_FET.lib:IRS2186'
- 'Driver_FET.lib:LM5109ASD'
- 'Driver_FET.lib:LM5109MA'
- 'MCU_Microchip_ATmega.lib:ATmega8L-8AU'
- 'MCU_Microchip_ATmega.lib:ATmega8L-8MU'
- 'MCU_Microchip_ATmega.lib:ATmega8L-8PU'
- 'MCU_Microchip_ATmega.lib:ATxmega256A3B-MH'
- 'MCU_Microchip_ATmega.lib:ATxmega256A3BU-AU'
- 'MCU_Microchip_ATmega.lib:ATxmega256A3BU-MH'
- 'MCU_Microchip_ATmega.lib:ATxmega64B1-AU'
- 'MCU_Microchip_ATmega.lib:ATxmega64B1-CU'
- 'MCU_Microchip_ATmega.lib:ATxmega64B3-AU'
- 'MCU_Microchip_ATmega.lib:ATxmega64B3-MH'
- 'Memory_Flash.lib:M29W004'
- 'Reference_Voltage.lib:TLE2426xLP'
- 'Sensor_Magnetic.lib:AS5045B'
- 'Sensor_Magnetic.lib:AS5048A'
- 'Sensor_Magnetic.lib:AS5048B'
- 'Sensor_Magnetic.lib:AS5055A'

Symbols where only NC pins have been touched (most likely moved onto the part outline. This changes should definetely be ok):
- 'Driver_FET.lib:EL7202CN'
- 'Driver_FET.lib:EL7212CN'
- 'Driver_FET.lib:EL7222CN'
- 'Driver_FET.lib:HIP2100_DFN'
- 'Driver_FET.lib:HIP2100_QFN'
- 'Driver_FET.lib:ICL7667'
- 'Driver_FET.lib:IR2110'
- 'Driver_FET.lib:LTC4440EMS8'
- 'Driver_FET.lib:MCP1415'
- 'Driver_FET.lib:MCP1415R'
- 'Driver_FET.lib:MCP1416'
- 'Driver_FET.lib:MCP1416R'
- 'Driver_FET.lib:MIC4426'
- 'Driver_FET.lib:MIC4427'
- 'Driver_FET.lib:MIC4428'

Ratfink, 2018-08-31:
Driver_FET.lib:IR2106 wasn't actually deleted, it was turned into an alias of Driver_FET.lib:IR2101 in d54589d82740b3bf5face204da2b548645abeaae.

poeschlr, 2018-08-31:
Oh. Hm that means my script needs a bit more power.
(Might also need to check for aliases that are no longer aliases.)

antoniovazquezblanco, 2018-10-04:
Release is comming. How are we doing with this? What needs to be done?

poeschlr, 2018-10-04:
Run the testscript i wrote, ask over at the mailing list if the changes are within acceptable limits. If yes tag the current master with 5.0.1. If not no new tag should be made in my opineon. But the packagers might not like it that way as they would need to change their scripts to still get the tag for 5.0.0 for the libraries.

antoniovazquezblanco, 2018-10-04:
Worst case scenario the tag for 5.0.1 can be set to point to the 5.0.0 commit in order to avoid changes for packagers. Little bit better scenario we can point to the previous commit that first breaks compatibility.

evanshultz, 2018-10-08:
So what is going to the included with 5.0.1? It's supposed to be tagged in a matter of hours (https://lists.launchpad.net/kicad-developers/msg37819.html).

I know you're going to check out the delta changes, Rene.

The only outstanding PR I see as important to merge is https://github.com/KiCad/kicad-symbols/pull/1044 since a user reported an error in the library and destroyed a part due to it. Otherwise, and since that one isn't critical, merging it and tagging 5.0.1 or leaving 5.0.0 is OK with me.

poeschlr, 2018-10-08:
Fix merged

poeschlr, 2018-10-08:
I ran my test script again. A few more symbols now have dangerous changes. (Main offender is the ATmega lib. Here the original symbols where nearly unsuitably large.)

Full report file: [report.txt](https://github.com/KiCad/kicad-symbols/files/2457610/report.txt)
There are 1183 changed symbols that are still compatible with earlier releases
There are 355 new symbols

Now a list of the incompatible changes:

Pins have been moved, renumbered or removed in the following symbols:
 - 'Connector.lib:USB_C_Plug' (#1005, Stack power pins -> netlist should stay unchanged in most cases)
 - 'Connector.lib:USB_C_Receptacle' (#1005, Stack power pins -> netlist should stay unchanged in most cases)
 - 'Driver_FET.lib:EL7202CN' (#891)
 - 'Driver_FET.lib:MC33152' alias of EL7202CN (#891)
 - 'Driver_FET.lib:MC34152' alias of EL7202CN (#891)
 - 'Driver_FET.lib:EL7212CN' (#891)
 - 'Driver_FET.lib:EL7222CN' (#891)
 - 'Driver_FET.lib:FAN7371' (#807)
 - 'Driver_FET.lib:FAN7842' (#807)
 - 'Driver_FET.lib:ICL7667' (#891)
 - 'Driver_FET.lib:IRS2186' (#807)
 - 'Driver_FET.lib:LM5109ASD'' (#807)
 - 'Driver_FET.lib:LM5109BSD' alias of LM5109ASD' (#807)
 - 'Driver_FET.lib:LM5109MA' (#807)
 - 'Driver_FET.lib:LM5109AMA' alias of LM5109MA (#807)
 - 'Driver_FET.lib:LM5109BMA' alias of LM5109MA (#807)
 - 'Driver_FET.lib:MIC4426' (#891)
 - 'Driver_FET.lib:MIC4427' (#891)
 - 'Driver_FET.lib:MIC4428' (#891)
 - 'MCU_Microchip_ATmega.lib:ATmega8L-8AU' (#774)
 - 'MCU_Microchip_ATmega.lib:ATmega8-16AU' alias of ATmega8L-8AU (#774)
 - 'MCU_Microchip_ATmega.lib:ATmega8A-AU' alias of ATmega8L-8AU (#774)
 - 'MCU_Microchip_ATmega.lib:ATmega8L-8MU' (#774)
 - 'MCU_Microchip_ATmega.lib:ATmega8-16MU' alias of ATmega8L-8MU (#774)
 - 'MCU_Microchip_ATmega.lib:ATmega8A-MU' alias of ATmega8L-8MU (#774)
 - 'MCU_Microchip_ATmega.lib:ATmega8L-8PU' (#774)
 - 'MCU_Microchip_ATmega.lib:ATmega8-16PU' alias of ATmega8L-8PU (#774)
 - 'MCU_Microchip_ATmega.lib:ATmega8A-PU' alias of ATmega8L-8PU (#774)
 - 'MCU_Microchip_ATmega.lib:ATmega8U2-AU' (#1013)
 - 'MCU_Microchip_ATmega.lib:ATmega16U2-AU' alias of ATmega8U2-AU (#1013)
 - 'MCU_Microchip_ATmega.lib:ATmega32U2-AU' alias of ATmega8U2-AU (#1013)
 - 'MCU_Microchip_ATmega.lib:ATxmega256A3B-MH' (#774)
 - 'MCU_Microchip_ATmega.lib:ATxmega256A3BU-AU' (#774)
 - 'MCU_Microchip_ATmega.lib:ATxmega256A3BU-MH' (#774)
 - 'MCU_Microchip_ATmega.lib:ATxmega64B1-AU' (#774)
 - 'MCU_Microchip_ATmega.lib:ATxmega128B1-AU' alias of ATxmega64B1-AU (#774)
 - 'MCU_Microchip_ATmega.lib:ATxmega64B1-CU' (#774)
 - 'MCU_Microchip_ATmega.lib:ATxmega128B1-CU' alias of ATxmega64B1-CU (#774)
 - 'MCU_Microchip_ATmega.lib:ATxmega64B3-AU' (#774)
 - 'MCU_Microchip_ATmega.lib:ATxmega128B3-AU' alias of ATxmega64B3-AU (#774)
 - 'MCU_Microchip_ATmega.lib:ATxmega64B3-MH' (#774)
 - 'MCU_Microchip_ATmega.lib:ATxmega128B3-MH' alias of ATxmega64B3-MH (#774)
 - 'MCU_Microchip_PIC16.lib:PIC16F871-IPT' (Might be a false positive. Can not find a commit that changed this symbol since v5.0.0)
 - 'Memory_Flash.lib:M29W004' (#928)
 - 'Power_Protection.lib:PESD3V3L4UF' (#882)
 - 'Power_Protection.lib:PESD5V0L4UF' alias of PESD3V3L4UF (#882)
 - 'Power_Protection.lib:PESD3V3L4UG' (#882)
 - 'Power_Protection.lib:PESD5V0L4UG' alias of PESD3V3L4UG (#882)
 - 'Power_Protection.lib:PESD3V3L4UW' (#882)
 - 'Power_Protection.lib:PESD5V0L4UW' alias of PESD3V3L4UW (#882)
 - 'Reference_Voltage.lib:MCP1525-TO' (#1044, Bugfix wrong pin numbers)
 - 'Reference_Voltage.lib:MCP1541-TO' alias of MCP1525-TO (#1044, Bugfix wrong pin numbers)
 - 'Reference_Voltage.lib:TLE2426xLP' (#686)
 - 'Sensor_Magnetic.lib:AS5045B' (#823)
 - 'Sensor_Magnetic.lib:AS5048A' (#823)
 - 'Sensor_Magnetic.lib:AS5048B' (#823)
 - 'Sensor_Magnetic.lib:AS5055A' (#823)
 - 'Sensor_Magnetic.lib:AS5050A' alias of AS5055A (#823)

Normal pins ok but NC pins have been moved, renumbered or removed in the following symbol:
 - 'Driver_FET.lib:HIP2100_DFN'
[... 19 lines cut ...]
Renamed (bugfix, added correct suffix to indicate which package the symbol uses. PR: https://github.com/KiCad/kicad-symbols/pull/826):
 - 'Regulator_Linear.lib:MIC5219-3.3' was an alias of AP131-15
 - 'Regulator_Linear.lib:MIC5219-2.5' was an alias of AP131-15
 - 'Regulator_Linear.lib:MIC5219-2.6' was an alias of AP131-15
 - 'Regulator_Linear.lib:MIC5219-2.7' was an alias of AP131-15
 - 'Regulator_Linear.lib:MIC5219-2.8' was an alias of AP131-15
 - 'Regulator_Linear.lib:MIC5219-2.9' was an alias of AP131-15
 - 'Regulator_Linear.lib:MIC5219-3.1' was an alias of AP131-15
 - 'Regulator_Linear.lib:MIC5219-3.0' was an alias of AP131-15
 - 'Regulator_Linear.lib:MIC5219-3.6' was an alias of AP131-15
 - 'Regulator_Linear.lib:MIC5219-5.0' was an alias of AP131-15
 - 'Regulator_Linear.lib:MIC5205-2.5' was an alias of AP131-15
 - 'Regulator_Linear.lib:MIC5205-2.7' was an alias of AP131-15
 - 'Regulator_Linear.lib:MIC5205-2.8' was an alias of AP131-15
 - 'Regulator_Linear.lib:MIC5205-2.85' was an alias of AP131-15
 - 'Regulator_Linear.lib:MIC5205-2.9' was an alias of AP131-15
 - 'Regulator_Linear.lib:MIC5205-3.0' was an alias of AP131-15
 - 'Regulator_Linear.lib:MIC5205-3.1' was an alias of AP131-15
 - 'Regulator_Linear.lib:MIC5205-3.2' was an alias of AP131-15
 - 'Regulator_Linear.lib:MIC5205-3.3' was an alias of AP131-15
 - 'Regulator_Linear.lib:MIC5205-3.6' was an alias of AP131-15
 - 'Regulator_Linear.lib:MIC5205-3.8' was an alias of AP131-15
 - 'Regulator_Linear.lib:MIC5205-4.0' was an alias of AP131-15
 - 'Regulator_Linear.lib:MIC5205-5.0' was an alias of AP131-15
 - 'Regulator_Linear.lib:MIC5205-ADJ'

poeschlr, 2018-10-08:
Wayne gave the ok so i will tag the current state of the lib as 5.0.1.
See: https://lists.launchpad.net/kicad-developers/msg37835.html

closed 2018-10-09 by poeschlr