KiCad/kicad-symbols #861: More consistent naming for symbols in Device.lib

fauxpark, 2018-08-30:
An offshoot of https://github.com/KiCad/kicad-symbols/pull/831#issuecomment-414030052.

I see some issues are being prebooked for v6, so I thought I should add this list of naming tweaks:

```
Amperemeter_AC -> Ammeter_AC
Amperemeter_DC -> Ammeter_DC
CP1 -> CP_Alt
CP1_Small -> CP_Small_Alt
CTRIM -> C_Trim
CTRIM_DIF C_Trim_Diff
DIAC_ALT -> DIAC_Alt
D_ALT -> D_Alt
D_Capacitance_ALT -> D_Capacitance_Alt
D_Photo_ALT -> D_Photo_Alt
D_Radiation_ALT -> D_Radiation_Alt
D_Schottky_Small_ALT -> D_Schottky_Small_Alt
D_Schottky_x2_ACom_AKK -> D_Schottky_Dual_CommonAnode_AKK
D_Schottky_x2_ACom_KAK -> D_Schottky_Dual_CommonAnode_KAK
D_Schottky_x2_ACom_KKA -> D_Schottky_Dual_CommonAnode_KKA
D_Schottky_x2_KCom_AAK -> D_Schottky_Dual_CommonCathode_AAK
D_Schottky_x2_KCom_AKA -> D_Schottky_Dual_CommonCathode_AKA
D_Schottky_x2_KCom_KAA -> D_Schottky_Dual_CommonCathode_KAA
D_Schottky_x2_Serial_ACK -> D_Schottky_Dual_Serial_ACK
D_Schottky_x2_Serial_AKC -> D_Schottky_Dual_Serial_AKC
D_Schottky_x2_Serial_CAK -> D_Schottky_Dual_Serial_CAK
D_Schottky_x2_Serial_CKA -> D_Schottky_Dual_Serial_CKA
D_Schottky_x2_Serial_KAC -> D_Schottky_Dual_Serial_KAC
D_Schottky_x2_Serial_KCA -> D_Schottky_Dual_Serial_KCA
D_Small_ALT -> D_Small_Alt
D_TVS_ALT -> D_TVS_Alt
D_TVS_x2_AAC -> D_TVS_Dual_AAC
D_TVS_x2_ACA -> D_TVS_Dual_ACA
D_TVS_x2_CAA -> D_TVS_Dual_CAA
D_Temperature_Dependent_ALT -> D_Temperature_Dependent_Alt
D_Tunnel_ALT -> D_Tunnel_Alt
D_Unitunnel_ALT -> D_Unitunnel_Alt
D_Zener_ALT -> D_Zener_Alt
D_Zener_Small_ALT -> D_Zener_Small_Alt
D_x2_ACom_AKK -> D_Dual_CommonAnode_AKK
D_x2_ACom_KAK -> D_Dual_CommonAnode_KAK
D_x2_ACom_KKA -> D_Dual_CommonAnode_KKA
D_x2_KCom_AAK -> D_Dual_CommonCathode_AAK
D_x2_KCom_AKA -> D_Dual_CommonCathode_AKA
D_x2_KCom_KAA -> D_Dual_CommonCathode_KAA
D_x2_Serial_ACK -> D_Dual_Serial_ACK
D_x2_Serial_AKC -> D_Dual_Serial_AKC
D_x2_Serial_CAK -> D_Dual_Serial_CAK
D_x2_Serial_CKA -> D_Dual_Serial_CKA
D_x2_Serial_KAC -> D_Dual_Serial_KAC
D_x2_Serial_KCA -> D_Dual_Serial_KCA
Delay_Line -> DelayLine
Electromagnetic_Actor -> ElectromagneticActor
Ferrite_Bead -> FerriteBead
Ferrite_Bead_Small -> FerriteBead_Small
Frequency_Counter -> FrequencyCounter
Hall_Generator -> HallGenerator
LED_ALT -> LED_Alt
LED_CRGB -> LED_KRGB
LED_Dual_2pin -> LED_Dual_Bidirectional
LED_Dual_AAC -> LED_Dual_AAK
LED_Dual_AACC -> LED_Dual_AAKK
LED_Dual_ACA -> LED_Dual_AKA
LED_Dual_ACAC -> LED_Dual_AKAK
LED_Dual_CAC -> LED_Dual_KAK
LED_Dual_CCA -> LED_Dual_KKA
LED_PAD -> LED_Pad
LED_RCBG -> LED_RKBG
LED_RCGB -> LED_RKGB
LED_Series_PAD -> LED_Series_Pad
LED_Small_ALT -> LED_Small_Alt
LTRIM -> L_Trim
Laserdiode_1A3C -> Laserdiode_1A3K
Laserdiode_1C2A -> Laserdiode_1K2A
Laserdiode_M_TYPE -> Laserdiode_MType
Laserdiode_N_TYPE -> Laserdiode_NType
Laserdiode_P_TYPE -> Laserdiode_PType
MEMRISTOR -> Memristor
Net-Tie_2 -> NetTie_2
Net-Tie_3 -> NetTie_3
Net-Tie_3_Tee -> NetTie_3_Tee
Net-Tie_4 -> NetTie_4
Net-Tie_4_Cross -> NetTie_4_Cross
Peltier_Element -> PeltierElement
Q_DUAL_NMOS_G1S2G2D2S1D1 -> Q_Dual_NMOS_G1S2G2D2S1D1
Q_DUAL_NPN_C2C1E1E2 -> Q_Dual_NPN_C2C1E1E2
Q_DUAL_NPN_NPN_E1B1C2E2B2C1 -> Q_Dual_NPN_NPN_E1B1C2E2B2C1
Q_DUAL_NPN_PNP_E1B1C2E2B2C1 -> Q_Dual_NPN_PNP_E1B1C2E2B2C1
Q_DUAL_PMOS_G1S2G2D2S1D1 -> Q_Dual_PMOS_G1S2G2D2S1D1
Q_DUAL_PNP_C2C1E1E2 -> Q_Dual_PNP_C2C1E1E2
Q_DUAL_PNP_PNP_C1B1B2C2E2E1 -> Q_Dual_PNP_PNP_C1B1B2C2E2E1
Q_DUAL_PNP_PNP_E1B1C2E2B2C1 -> Q_Dual_PNP_PNP_E1B1C2E2B2C1
RF_Shield_One_Piece -> RF_Shield_OnePiece
RF_Shield_Two_Pieces -> RF_Shield_TwoPieces
RTRIM -> R_Trim
R_PHOTO -> R_Photo
R_POT -> R_Pot
R_POT_Dual -> R_Pot_Dual
R_POT_Dual_Separate -> R_POT_Dual_Separate
R_POT_TRIM -> R_Pot_Trim
R_POT_TRIM_US -> R_Pot_Trim_US
R_POT_US -> R_Pot_US
Rotary_Encoder -> RotaryEncoder
Rotary_Encoder_Switch -> RotaryEncoder_Switch
SPARK_GAP -> SparkGap
Thermocouple_ALT -> Thermocouple_Alt
Transformer_AUDIO -> Transformer_Audio
Voltage_Divider -> VoltageDivider
Voltage_Divider_CenterPin1 -> VoltageDivider_CenterPin1
Voltage_Divider_CenterPin3 -> VoltageDivider_CenterPin3
```

 - Words that are in all caps but not an acronym eg. `MEMRISTOR`, are changed to be Capitalized
 - I interpret the underscore in symbol names to delimit the different properties of the symbol (NPN vs PNP transistors for example), so things like "Rotary_Encoder" and "Voltage_Divider" should be concatenated with no underscore
 - `CP1` (polarized capacitor with curved cathode) should conform to the defacto standard suffix of `_ALT` to denote an alternate symbol (perhaps it should be `CP_CurvedCathode` instead?)
 - Dual diodes are changed from `x2` to `Dual` for clarity
 - The `C` in dual/RGB LED symbol names is changed to `K` in keeping with other diode symbols and to prevent confusion with some dual diodes which use `C` to mean the "center" pin

poeschlr, 2018-08-31:
This will have to wait till we start the version 6 development as changing symbol names (especially at that scale in this library) would likely break user projects.

It is however a good idea to already think about good naming schemes and use it as the basis for new additions.

matthijskooijman, 2019-02-20:
Some of the suggested renames change "x2" to "Dual". However, I can't find any mention of "Dual" in the KLC. The only thing I see is [S2.1.7](http://kicad-pcb.org/libraries/klc/S2.1/):

> Indicate quantity of elements for symbol arrays (e.g. resistor array with 8 elements - Resistor_x8)

Whose example would suggest x2 is actually ok (though the policy itself just says "indicate quantity", so "Dual" would also satisfy that). Perhaps the KLC needs clarification here?

evanshultz, 2019-02-20:
I agree that more clarity is needed. Relatedly, https://github.com/KiCad/kicad-symbols/issues/1406 mentions a similar situation for resistor arrays.

fauxpark, 2019-02-20:
I don't see those symbols as arrays, especially the `_Serial` diodes because the two diodes are not facing the same direction. Electrically, you could look at them as being arrayed, but physically they are kind of laid out in a Y shape:
[image]

And (in my admittedly limited experience) you would find these in SOT packages which don't exactly look like arrays of diodes. So I think `_Dual` is the most apt description of these symbols.

evanshultz, 2019-02-21:
I believe the word "array" is used to mean unconnected multiple elements in the same package. So a single pair of diodes, just like shown above, is not an array. But two of them (one of the above in each unit) are an array. The "x2" symbols have two units and so are an array. At least that's what I took "array" to mean.

fauxpark, 2019-02-21:
None of the Device.lib symbols marked `x2` have multiple units.

fauxpark, 2019-03-16:
@poeschlr @evanshultz now that 5.1 is out, is it time for me to start work on this? Or still too early?

PR #1650 (2020-09-16): Fix naming convention for Device.lib
Fixes #861.

For v6, here we have a whole bunch of symbol renames for Device.lib for consistency and readability. I've broken it down into smaller commits so it's easier to digest.

---

All contributions to the kicad library must follow the [KiCad library convention](http://kicad-pcb.org/libraries/klc/)

Thanks for creating a pull request to contribute to the KiCad libraries! To speed up integration of your PR, please check the following items:

- [x] Provide a URL to a datasheet for the symbol(s) you are contributing
- [x] An example screenshot image is v [... cut]
closed 2020-09-16 by myfreescalewebpage