KiCad/kicad-symbols #871: Missing a wildcard logo symbol

5N44P, 2018-08-31:
Picture this scenario: 
I make a schematic. I finish the routing, then I put my logo and the kicad logo on the pcb with "Add footprints" 
At this point, i do some modifications to the schematic, and i press F8 to update my pcb. 
All the logos disappear because they aren't associated with a symbol in eeschema. 

Currently in the Graphics library i didn't found any symbol that would fit a generic logo footprint. 

I think that Graphic.lib should contain a symbol for this purpose. 

If there is one already, i'm sorry but i didn't found it

5N44P, 2018-08-31:
@evanshultz I already knew about the lock footprint option. 
That was not my issue. I was not asking for a way to fix my particular case. I was picturing an hypotetic scenario.
My issue is about the lack of consistency. 

If the symbol + footprint is the way to go, then there should be a symbol for that. 
If locking the footprint is the way to go, a part beeing very unconvenient, then the mounting holes symbols are not needed.

I think that this inconsistency should be solved by adding a logo symbol. 

I can also make a PR for that, but first i think that it is necessary to discuss it here.

evanshultz, 2018-09-01:
Thanks for the extra explanation.

Some mounting holes have a net connection and must be in the schematic. A logo that appears on the PCB does not have a net connection and does not need to have a schematic symbol. Similarly, a hole in the PCB that does not have any net connection does not need a symbol.

That is the dichotomy: If a part needs a net connection it must have a schematic symbol. Otherwise, it does not. And if the logo symbol is generic, it won't automatically convey its purpose to the user which makes it even less useful to me. Why require a symbol to be placed on a schematic when that work is not necessary?

Furthermore, CvPcb only includes symbols with pins. The logos in Graphics.lib do not have pins and so they cannot have a footprint association as far as I can tell. Am I wrong? (Then to even pick a footprint in CvPcb, the user must turn off all filtering as the symbols also have no footprint filters.)

So to your point, you don't want the logo in the PCB to disappear when a schematic update is pushed to the PCB. That's quite reasonable. Locking the footprint solves this issue and doesn't require any symbols to be made and managed.

I don't personally see the point of making symbols for this use case. But I'm only me and not everybody. It may be a valuable addition to the KiCad library for many people. You, however, can make your own symbol for your personal use in your own personal library if the logo is just for you.

5N44P, 2018-09-01:
> Some mounting holes have a net connection and must be in the schematic. A logo that appears on the PCB does not have a net connection and does not need to have a schematic symbol. Similarly, a hole in the PCB that does not have any net connection does not need a symbol.
That is the dichotomy: If a part needs a net connection it must have a schematic symbol. 

I understand that, but that's not always the case. See for example _Mechanical:MountingHole_. It doesn't have any pin but still allows you to associate a footprint to it.    
Then you have _Mechanical:MountingHole_Pad_ but that's a different symbol, and honestly it makes sense to me to have both.

> And if the logo symbol is generic, it won't automatically convey its purpose to the user which makes it even less useful to me.

We can name it something explicit such as "Pcb logo", "Pcb Art", "Pcb Silkscreen symbol", "Generic user logo" or something like this. 

> Locking the footprint solves this issue and doesn't require any symbols to be made and managed.

Yes, of course it does. But it's incovenient, intricate and overcomplicated. It also gives you warnings when you move around them. To me, this isn't a long term solution. 

> You, however, can make your own symbol for your personal use in your own personal library if the logo is just for you.

Yes i can, but i also care about the fate of kicad, and i think that this inconsistency should be solved. I'm raising this issue because i think that this is a serious discrepancy in the software, and we should do our best to solve it for everyone. 

>  But i do not really have an idea how this symbol could look like, what reference prefix to give to it, in what library to put it nor what it should be named.

To me, just a plain empty box would be descriptive enough.  
A reference such as __SYM?__ to me looks good, and i think it should go in the _Graphic_ lib, since it is only something graphic. The name is tougher. Above i've proposed some, they don't really convice me, but i think that it is a starting point.