KiCad/kicad-symbols #872: Missing symbols for magnetic sensors and number of other libraries

YordanYanakiev, 2018-08-31:
Hello.
There is missing symbols for some libraries in 5.x.

antoniovazquezblanco, 2018-09-18:
Ping @YordanYanakiev. Please, could you answer the questions above?

antoniovazquezblanco, 2018-11-08:
@YordanYanakiev ping!

closed 2018-11-22 by antoniovazquezblanco

evanshultz, 2018-11-22:
Just to try and wrap this up...

I don't think it's about schematic symbols (as originally reported) because https://github.com/KiCad/kicad-library/pull/1697/files are the file changes for an attempt to split up the sensor libs in the old library. Comparing with the current lib at https://github.com/KiCad/kicad-symbols/blob/master/Sensor_Magnetic.dcm, all old symbols are there.

If about footprints, while it's true that https://github.com/KiCad/kicad-footprints does not have a Sensor_Magnetic repo, all symbols in the current Sensor_Magnetic symbol library are associated with generic footprints that do exist in our footprint library.

So I don't see an issue with magnetic sensors at all and any other real issue can be surfaced later.