KiCad/kicad-symbols #895: "Global Symbol Fields" in Schematic Editor and "Local Symbol Fields" in Symbol Editor

DrBockel, 2018-09-05:
Having trouble with the symbol fields in symbol and schematic editor I would like to discuss the following:

1) In the Symbol Editor /Local Level I would like to use only "Local Fields" which is in my opinion are only the Fields "Reference","Description" and "Datasheet". 
No Footprint(s):
Different Footprints make me to set up different Symbols which is not necessary on this level. I will decide the footprint later in schematic or in pcb layout maybe even after checking what is available in the market or what can I handle in assembly. Definitly after setting up a schematic and not a symbol. Sometimes it is clear at the beginning but in many cases not.
No Value(s):
A value like 10k gives me no more information at this level. With details like temp rating, tolerance, voltage etc. I work on a schematic/global scope.

2) In the Schematic Editor / Global Level I use the "Global Fields" which are Value, Footprint, revised Datasheet(s), revised Description, Notes, Details like temperature, voltage etc. to make the thing work in theory :)
It is really nice that with 5.0. there is now an editor to enter/change these quick on globel/schematic level.

3)Next step in the workflow is to create a BOM out of that for example acc. to Octopart specs. 
Looking now for a part which has not only the technical specs but also the commercial, is available, has a lot of supplier etc. will need me to add manufacturer numbers. Here I start reading all the datasheets and compare "similar" parts at the distributors to decide for some manf#. Fill them in and ready is my first BOM. Ups part is obsolete...I need to find another one..... SO there is a need for a lot of custom fileds which I dont know and maybe change in future.

**Summary what I would like to have:
Local Fields in Symbol Editor - Only fields "Reference" ,"Desciption", "Datasheet". optional add everything you want
Global Fields in Schematic Editor - Only fields "Footprint", "Value" optional add everything you want except fields from Symbol Editor.
Editor for Symbol Fields in Symbol Libraries - open 
Editor for Symbol Fields in Schematic - missing features clear, add all fields or single fields**

Lookig forward to your comments.
BR Marcus

poeschlr, 2018-09-05:
I am not really sure you choose the correct place to ask this. We here only take care about the official library symbols. We can not change anything in the software it self. Requests for the software must be placed in the bugtracker on launchpad: https://bugs.launchpad.net/kicad

antoniovazquezblanco, 2018-11-08:
@DrBockel Given that we cannot handle your feature request can we close this issue? Have you tried the forums or launchpad bugtracker?

Thanks

DrBockel, 2018-11-18:
You can close it here.

closed 2018-11-19 by antoniovazquezblanco