KiCad/kicad-symbols #897: Need symbol for arduino mega

ajlaut, 2018-09-06:
The arduino uno symbol is very useful and better defined with DRC than that available via the Sparkfun/Ki-Cad librarie. It would be nice if the same were available for the arduino mega. The more the better!

evanshultz, 2018-09-06:
I assume the UNO symbols you're talking about these first two results here:
[image]

Good idea to have one for the MEGA. And others. Create the symbols and submit a PR.

antoniovazquezblanco, 2018-09-18:
@ajlaut we do not take symbol petitions in this bugtracker due to the high amount of work they generate. I fear this issue should be closed.

On the other hand we would be glad if you decided to contribute with a symbol for the Arduino MEGA component you are looking for.

Thank you!

closed 2018-09-18 by antoniovazquezblanco