KiCad/kicad-symbols #911: weekley release tags please

greyltc, 2018-09-10:
The `README.md` says that weekly releases can be found at https://kicad.github.io/symbols/

Can we get tags here that correspond to those weekly releases?

This would help me in packaging this for Arch Linux.

poeschlr, 2018-09-16:
These are not really releases but snapshots. Adding tags every week will clutter the github pages for the library repos. If you want a sort of nightly release then use the master branch directly as is done for the source code of kicad.

antoniovazquezblanco, 2018-11-08:
@greyltc Does this resolve your issues? Can this be closed?

greyltc, 2018-11-12:
Is there a place I can go to see the history of the snapshots?
I don't really understand the cluttering actually, I thought tags just show up here: https://github.com/KiCad/kicad-symbols/tags

evanshultz, 2019-02-22:
@poeschlr 
Ping.

poeschlr, 2019-02-25:
tags sadly also show up on the release pages. Not in the same size but they are still shown.

See: https://github.com/KiCad/kicad.github.io/releases -> this is where you get your already compressed snapshots from.

myfreescalewebpage, 2019-03-15:
@greyltc does it solves your issue ?

greyltc, 2019-03-15:
Yes!

closed 2019-03-15 by greyltc

myfreescalewebpage, 2019-03-15:
Thanks for the update!