KiCad/kicad-symbols #912: Crypto devices new library

antoniovazquezblanco, 2018-09-10:
I would like to provide a symbol for ATECC608A (http://ww1.microchip.com/downloads/en/DeviceDoc/40001977A.pdf), a Microchip CryptoAuthentication Device but I could not find a suitable library for the matter.

Any suggestions for a library name that can hold this device?

evanshultz, 2018-09-10:
Yeah, I don't see anything that really fits. Good question.

Microchip has this in a group of parts called "Security ICs".

Let's look a bit more. DigiKey groups it into "Specialized ICs" which isn't too helpful or unique. Mouser uses "Semiconductors / Security ICs / Authentication ICs". Arrow puts it under "Microcontrollers and Processors / Authenticators". Other vendors were similar.

To be a bit more broad than just crypto, what about `Security.lib`?

pointhi, 2018-09-10:
I thought about ```Security_Crypto.lib```. Another library name in the future would be for example ```Security_RNG.lib``` then.

Problematic with this approach would be the handling of chips including multiple features like additional secure storage (in the given chip included), or Trusted platform modules. Using a more generic name like ```Security.lib``` would prevent this issue.

antoniovazquezblanco, 2018-09-11:
I would go for the `Security.lib` for the time being but I've just remembered that we won't be able to rename or move components between "small" releases so a more specialized name would be a good idea...

antoniovazquezblanco, 2018-09-11:
PR opened in https://github.com/KiCad/kicad-symbols/pull/918.

Discusion will follow there.

closed 2018-09-11 by antoniovazquezblanco