KiCad/kicad-symbols #936: PR Travis CI Build error if datasheet field is filled

serkanxselcuk, 2018-09-15:
Is there any reason to leave datasheet field free while adding a new part into symbol libraries? I think it would be helpful for users.

poeschlr, 2018-09-16:
I am not sure what you mean here. Filling out the datasheet field is required by our rules. Travis should only complain if it is empty!
If you want it intentionally empty use "~" (this is only to be used for generic symbols)

poeschlr, 2018-09-16:
Ah i think i understand your confusion. Kicad has two places to store datasheets. one is in the symbol field (stored in the lib file) and one is in the symbol properties (or more precisely alias properties) that one is stored in the dcm fil. if the field in the lib file is empty then it will be filled with the entry of the dcm file.

The dcm file entry allows adding different datasheets to different aliases and is therefore the option choosen for the official lib.

serkanxselcuk, 2018-09-16:
> if the field in the lib file is empty then it will be filled with the entry of the dcm file.
Oh, I see. Thanks for quick respond. Then it is better solved then I thought.

closed 2018-09-16 by serkanxselcuk