KiCad/kicad-symbols #938: Electrical type of `Vss` pin of `STM8S003F` series microcontroller symbol seems wrong!

MostafaYaghmaei, 2018-09-16:
Hi,
I've installed the following version of `KiCad` in my `ubuntu 18.04`:
`Version: 5.0.0-fee4fd1~66~ubuntu18.04.1, release build`
I used the `KiCad` default symbol libraries to place `STM8S003F3P` into schematic. After getting error in ERC check, I found that the electrical type of `Vss` pin of the symbol is set to `power output` , while it's set to `power input` in any other micro controllers.

evanshultz, 2018-09-17:
Yup, seems wrong to me too. Would you mind submitting a PR to fix it?

evanshultz, 2018-09-18:
This may be a good chance to try. Would you like to make your first PR to correct this issue?

MostafaYaghmaei, 2018-09-19:
Yes, of course. I would like to submit my first PR. What should I do exactly?

evanshultz, 2018-09-19:
You have two options:
1. Use the GitHub web client.
2. Make the changes on your local computer and commit them. Then submit a new PR to this repo of the changes. (This is the preferred option for several reasons.)

See https://help.github.com/articles/creating-a-pull-request-from-a-fork/ for method 1 or https://github.com/PointCloudLibrary/pcl/wiki/A-step-by-step-guide-on-preparing-and-submitting-a-pull-request for method 2. This is only the very basics and the first seemingly relevant link I found, so as you mentioned learning above so I'll leave it to you to Google for more information. Come back here if you need help, but please search for answers first so we can focus on KiCad-specific topics here and not Git topics.

Thanks!

mstfyghm, 2018-09-19:
Thanks! :+1:

evanshultz, 2018-09-20:
Fixed!

closed 2018-09-20 by evanshultz