KiCad/kicad-symbols #939: What RefDes standard does KiCad use?

evanshultz, 2018-09-16:
What standard is KiCad using as the basis for the reference designators http://kicad-pcb.org/libraries/klc/S6.1/? Is there one?

IEEE 315 has been used for many items like symbol style, and it's mentioned at https://github.com/KiCad/kicad-symbols/pull/612. But https://github.com/KiCad/kicad-symbols/pull/907 updated based on EN 81346 (or IEC 81346). The first link also references using common industry standards.

Bending or breaking a standard for a good reason is fine with me, but is there a standard that KiCad falls back on?

poeschlr, 2018-09-17:
@SchrodingersGat i think you wrote that section of the KLC. could you comment on that as i don't know anything about this.

SchrodingersGat, 2018-09-17:
The list was originally adapted from the [Wikipedia](https://en.wikipedia.org/wiki/Reference_designator) page which loosely follows IEEE-315. It has never been formalized insofar as picking a single standard to follow.

serkanxselcuk, 2018-09-17:
Some letters of IEEE-315:1975 is kind to use in PCB designs, here is the link to that (see section 22 at page 81):
https://www.noao.edu/ets/Mechanical/Policies/ANSI%20Y32.2-1975.pdf

Thema is some confused, because people learn class designation letters in high schools and university, which are not completely implemented on standardization institutes. Some of them are taken from IEEE-315, others seems like IEC/EN 81346-2. This why common usage is different then standards and norms.

IEEE-315 | IEC/EN 81346 | Common usage in PCB design | Description
--- | --- | --- | ---
L | C | L | Inductive
B | M | M | Motor
P, J | X | J, X | Connector
Not allowed | N | ? | Housing/Enclosure

Designation MK was matching with microphone. Housing/Enclosure is not allowed to get any designation under IEEE-315, but we need a letter in KiCad to place it as component. I wanted to change letter to any known one in industrial usage. Because of that I suggested N for housing in #907 .

At the other side, KiCad makes it theoretic possible to design machine, electrical cabinets and cable plans. Maybe it would be interesting to think of implementation of some electrical component into libraries in cooperation with FreeCad for this usage like some other SW (Eagle, AutoCAD, WSCAD) does?

serkanxselcuk, 2018-09-22:
I think thema about RefDes will be opened frequently. What do you think about to insert a RefDes.csv file into footprint library pfad? So would it be able to develop (e.g. adding more examples to designator letter for clearifying what includes symbol) by contributors and reviewers can check as well.

serkanxselcuk, 2018-09-22:
@evanshultz Can we make changes on KLC S6.1?

I mean contributors can add "letters" or "example devices for letters". And reviewers check if new change on RefDes can be implemented.