KiCad/kicad-symbols #950: Transistor_FET: S5.1 and S5.2 errors

antoniovazquezblanco, 2018-09-18:
KLC Rules:
http://kicad-pcb.org/libraries/klc/S5.1/
http://kicad-pcb.org/libraries/klc/S5.2/

Splitting issue #28 I have found the following violations:
```
Checking symbol 'IRF40DM229':
  Violating S5.1
    For components with a single default footprint, footprint field is filled with valid footprint
    filename
    Specified footprint does not exist
     - Footprint file Package_DirectFET:DirectFET_MF was not found
Checking symbol 'IRF6718L2':
  Violating S5.1
    For components with a single default footprint, footprint field is filled with valid footprint
    filename
    Specified footprint does not exist
     - Footprint file Package_DirectFET:DirectFET_L6 was not found
checking ruleS5.2
checking ruleS5.1
Checking symbol 'IRF6802SD':
  Violating S5.1
    For components with a single default footprint, footprint field is filled with valid footprint
    filename
    Specified footprint does not exist
     - Footprint file Package_DirectFET:DirectFET_SA was not found
checking ruleS5.2
checking ruleS5.1
Checking symbol 'IRF6892S':
  Violating S5.1
    For components with a single default footprint, footprint field is filled with valid footprint
    filename
    Specified footprint does not exist
     - Footprint file Package_DirectFET:DirectFET_S3C was not found
Checking symbol 'IRF7665S2':
  Violating S5.1
    For components with a single default footprint, footprint field is filled with valid footprint
    filename
    Specified footprint does not exist
     - Footprint file Package_DirectFET:DirectFET_SB was not found
Checking symbol 'IRF9394M':
  Violating S5.1
    For components with a single default footprint, footprint field is filled with valid footprint
    filename
    Specified footprint does not exist
     - Footprint file Package_DirectFET:DirectFET_MC was not found
Checking symbol 'IRL6283M':
  Violating S5.1
    For components with a single default footprint, footprint field is filled with valid footprint
    filename
    Specified footprint does not exist
     - Footprint file Package_DirectFET:DirectFET_MD was not found
```

evanshultz, 2018-09-18:
https://github.com/KiCad/kicad-footprints/issues/608

evanshultz, 2019-09-26:
All DirectFET footprints used by symbols are now present in the library.

closed 2019-09-26 by evanshultz