KiCad/kicad-symbols #953: Timer: S5.1 and S5.2 errors

antoniovazquezblanco, 2018-09-18:
KLC Rules:
http://kicad-pcb.org/libraries/klc/S5.1/
http://kicad-pcb.org/libraries/klc/S5.2/

Splitting issue #28 I have found the following violations:
```
Checking symbol '8284':
  Violating S5.1
    For components with a single default footprint, footprint field is filled with valid footprint
    filename
    Footprint filter 'PDIP*W7.62mm*' does not match the footprint 'Package_DIP:DIP-18_W7.62mm' set for this symbol.
     - could not match 'DIP-18_W7.62mm' against filter 'PDIP*W7.62mm*'
     - could not match 'Package_DIP:DIP-18_W7.62mm' against filter 'PDIP*W7.62mm*'
    Symbol has a footprint defined in the footprint field, but several (2) footprint filters set. If the symbol is for a single default footprint, remove the surplus filters. If the symbol is meant for multiple different footprints, empty the footprint field.
checking ruleS5.2
checking ruleS5.1
Checking symbol '82C54':
  Violating S5.1
    For components with a single default footprint, footprint field is filled with valid footprint
    filename
    Footprint filter 'PDIP*W15.24mm*' does not match the footprint 'Package_DIP:DIP-24_W15.24mm' set for this symbol.
     - could not match 'DIP-24_W15.24mm' against filter 'PDIP*W15.24mm*'
     - could not match 'Package_DIP:DIP-24_W15.24mm' against filter 'PDIP*W15.24mm*'
    Symbol has a footprint defined in the footprint field, but several (2) footprint filters set. If the symbol is for a single default footprint, remove the surplus filters. If the symbol is meant for multiple different footprints, empty the footprint field.
```

serkanxselcuk, 2018-09-18:
Hi @antoniovazquezblanco , 

>Footprint filter 'PDIP*W7.62mm*' does not match the footprint 'Package_DIP:DIP-18_W7.62mm' set for this symbol.
     - could not match 'DIP-18_W7.62mm' against filter 'PDIP*W7.62mm*'
     - could not match 'Package_DIP:DIP-18_W7.62mm' against filter 'PDIP*W7.62mm*'

You filtered `PDIP*W7.62mm*` and set footprint as `DIP-18_W7.62mm`, which cannot be found using filter. KLC shows this as error. I would try to set filter `*DIP*W7.62mm*`' or `DIP*W7.62mm*`.

serkanxselcuk, 2018-09-18:
Oh, I see. I thought you try to import new component. :) Thank you for focus on old issues.

serkanxselcuk, 2020-03-11:
@antoniovazquezblanco 
I remember my first message here :)