KiCad/kicad-symbols #959: Regulator_Switching: S5.1 and S5.2 errors

antoniovazquezblanco, 2018-09-18:
KLC Rules:
http://kicad-pcb.org/libraries/klc/S5.1/
http://kicad-pcb.org/libraries/klc/S5.2/

Splitting issue #28 I have found the following violations:
```
Checking symbol 'KA5H0265RCTU':
  Violating S5.1
    For components with a single default footprint, footprint field is filled with valid footprint
    filename
    Specified footprint does not exist
     - Footprint file Package_TO_SOT_THT:ONSemi_TO-220-5L was not found
Checking symbol 'KA5H0265RCYDTU':
  Violating S5.1
    For components with a single default footprint, footprint field is filled with valid footprint
    filename
    Specified footprint does not exist
     - Footprint file Package_TO_SOT_THT:ONSemi_TO-220-5L_Forming was not found
Checking symbol 'KA5M0265RYDTU':
  Violating S5.1
    For components with a single default footprint, footprint field is filled with valid footprint
    filename
    Specified footprint does not exist
     - Footprint file Package_TO_SOT_THT:ONSemi_TO-220F-4L_Forming was not found
Checking symbol 'LT8610':
  Violating S5.1
    For components with a single default footprint, footprint field is filled with valid footprint
    filename
    Specified footprint does not exist
     - Footprint file Package_SO:LTC_MSOP-16-1EP_3x4mm_P0.5mm_EP2.15x3.26mm was not found
checking ruleS5.2
checking ruleS5.1
Checking symbol 'LT8610AC':
  Violating S5.1
    For components with a single default footprint, footprint field is filled with valid footprint
    filename
    Specified footprint does not exist
     - Footprint file Package_SO:LTC_MSOP-16-1EP_3x4mm_P0.5mm_EP2.15x3.26mm was not found
Checking symbol 'LTC3525':
  Violating S5.1
    For components with a single default footprint, footprint field is filled with valid footprint
    filename
    Symbol has a footprint defined in the footprint field, but several (2) footprint filters set. If the symbol is for a single default footprint, remove the surplus filters. If the symbol is meant for multiple different footprints, empty the footprint field.
Checking symbol 'TPS82130':
  Violating S5.1
    For components with a single default footprint, footprint field is filled with valid footprint
    filename
    Specified footprint does not exist
     - Footprint file Package_LGA:Texas_MicroSiP-8-1EP_2.8x3.0mm_P0.65mm_EP1.1x1.9mm_SMD_ThermalVias was not found
```

rs-development, 2018-12-24:
Is this still open i am not getting this errors?

ahatstat, 2019-04-13:
This appears to still be open. I tried to use TPS82130 symbol in a design.  Package_LGA:Texas_MicroSiP-8-1EP_2.8x3.0mm_P0.65mm_EP1.1x1.9mm_SMD_ThermalVias is not in the footprint library in version 5.02.