KiCad/kicad-symbols #975: MCU_Module: S5.1 and S5.2 errors

antoniovazquezblanco, 2018-09-18:
KLC Rules:
http://kicad-pcb.org/libraries/klc/S5.1/
http://kicad-pcb.org/libraries/klc/S5.2/

Splitting issue #28 I have found the following violations:

```
Checking symbol 'CHIP':
  Violating S5.1
    For components with a single default footprint, footprint field is filled with valid footprint
    filename
    Specified footprint does not exist
     - Footprint file Module:MODULE_NEXTTHINGCO_CHIP was not found
checking ruleS5.2
checking ruleS5.1
Checking symbol 'CHIP-PRO':
  Violating S5.1
    For components with a single default footprint, footprint field is filled with valid footprint
    filename
    Specified footprint does not exist
     - Footprint file Module:MODULE_NEXTTHINGCO_CHIPPRO was not found
```

myfreescalewebpage, 2018-12-06:
Hi,
I have proposed these symbol when working with old Kicad 4 libraries but the footprint was still pending when going to Kicad 5.
I'm working to pull request again the footprints on the new repository with updated KLC rules.
Cheers
Joel

poeschlr, 2018-12-06:
Could you add a link to the old pull request here? This would help us if you for some reason can not finish this.

myfreescalewebpage, 2018-12-06:
Well, I have deleted my old forks on github, actually the footprint is on my personal computer only.

myfreescalewebpage, 2018-12-08:
Pull requests done : https://github.com/KiCad/kicad-footprints/pull/1156 and https://github.com/KiCad/kicad-symbols/pull/1240

antoniovazquezblanco, 2018-12-10:
Thanks! It is nice to see a fix!