KiCad/kicad-symbols #978: Power_Protection: S5.1 and S5.2 errors

antoniovazquezblanco, 2018-09-18:
KLC Rules:
http://kicad-pcb.org/libraries/klc/S5.1/
http://kicad-pcb.org/libraries/klc/S5.2/

Splitting issue #28 I have found the following violations:

```
Checking symbol 'CM1213A-01SO':
  Violating S5.1
    For components with a single default footprint, footprint field is filled with valid footprint
    filename
    Footprint filter '*TO*236*' does not match the footprint 'Package_TO_SOT_SMD:SOT-23' set for this symbol.
     - could not match 'SOT-23' against filter '*TO*236*'
     - could not match 'Package_TO_SOT_SMD:SOT-23' against filter '*TO*236*'
    Symbol has a footprint defined in the footprint field, but several (2) footprint filters set. If the symbol is for a single default footprint, remove the surplus filters. If the symbol is meant for multiple different footprints, empty the footprint field.

Checking symbol 'NUP2202':
  Violating S5.1
    For components with a single default footprint, footprint field is filled with valid footprint
    filename
    Footprint filter '*SC*88*' does not match the footprint 'Package_TO_SOT_SMD:SOT-363_SC-70-6' set for this symbol.
     - could not match 'SOT-363_SC-70-6' against filter '*SC*88*'
     - could not match 'Package_TO_SOT_SMD:SOT-363_SC-70-6' against filter '*SC*88*'
    Symbol has a footprint defined in the footprint field, but several (3) footprint filters set. If the symbol is for a single default footprint, remove the surplus filters. If the symbol is meant for multiple different footprints, empty the footprint field.

Checking symbol 'SP0502BAJT':
  Violating S5.1
    For components with a single default footprint, footprint field is filled with valid footprint
    filename
    Footprint filter 'SC?70*' does not match the footprint 'Package_TO_SOT_SMD:SOT-323_SC-70' set for this symbol.
     - could not match 'SOT-323_SC-70' against filter 'SC?70*'
     - could not match 'Package_TO_SOT_SMD:SOT-323_SC-70' against filter 'SC?70*'

Checking symbol 'SP0504BAJT':
  Violating S5.1
    For components with a single default footprint, footprint field is filled with valid footprint
    filename
    Footprint filter 'SC?70*' does not match the footprint 'Package_TO_SOT_SMD:SOT-353_SC-70-5' set for this symbol.
     - could not match 'SOT-353_SC-70-5' against filter 'SC?70*'
     - could not match 'Package_TO_SOT_SMD:SOT-353_SC-70-5' against filter 'SC?70*'

Checking symbol 'SP0505BAJT':
  Violating S5.1
    For components with a single default footprint, footprint field is filled with valid footprint
    filename
    Footprint filter 'SC?70*' does not match the footprint 'Package_TO_SOT_SMD:SOT-363_SC-70-6' set for this symbol.
     - could not match 'SOT-363_SC-70-6' against filter 'SC?70*'
     - could not match 'Package_TO_SOT_SMD:SOT-363_SC-70-6' against filter 'SC?70*'
```

evanshultz, 2018-09-19:
At least some fixed in https://github.com/KiCad/kicad-symbols/pull/882.

antoniovazquezblanco, 2018-09-19:
It seems that the work from @evanshultz was there before this PR so I will close this in favor of #882.

Thank you anyways @serkanxselcuk

antoniovazquezblanco, 2018-09-19:
Closed by accident