KiCad/kicad-symbols #989: USB_B_Micro in Connector.lib has wrong electrical type on GND

r4ptor, 2018-09-20:
The symbol USB_B_Micro in Connector.lib has the GND pin labeled as power output.

Since it's a ground pin this is incorrect. It should be labeled as passive.

This can lead to problems with the ERC in the schematic.
In my case a design with a TI TPS60501 led to the error that two power outputs are connected. 

So in Connector.lib for the section of USB_B_Micro

`X GND 5 0 -400 100 U 50 50 1 1 w`

should be replaced with

`X GND 5 0 -400 100 U 50 50 1 1 P`

antoniovazquezblanco, 2018-09-21:
Hi @r4ptor !

I believe the electrical type is right. Let me explain:

Imagine you have a host USB machine that defines a potential for GND. When you plug a cable into your USB B micro connector, then the host machine is trying to impose a potential in your GND pin, thus acting as "power source". If your circuit has another GND power source you need to use a fuse or a net tie to solve this problem. If you are aware of that problem you may also ignore that issue in the DRC.

I hope this clarifies a little bit the reasoning behind the scenes. Does this help you?

r4ptor, 2018-09-21:
Hi @antoniovazquezblanco,

thanks for the explanation, makes sense for me. 

Closing the issue.

closed 2018-09-21 by r4ptor

antoniovazquezblanco, 2018-09-21:
Thank you @r4ptor!