KiCad/kicad-symbols #992: STM32 symbols should have stacked power inputs

DanSGiesbrecht, 2018-09-22:
Many STM32 symbols have many power pins which would benefit from being **stacked to reduce symbol width**:

For example, **STM32F722ICKx**:
[image]

There are numerous redundant power pins:
[image]

evanshultz, 2018-09-22:
https://github.com/KiCad/kicad-symbols/pull/305 merged the symbols and according to our guidelines (KLC updates are severely lagging these days) if the datasheet says to place a decoupling cap on a pin then the pin should not be stacked.

The PR references indicates this is the case for each pin pair, but I didn't check this myself and it may not apply to all members of the family.

Also, if each pin pair is supposed to get a cap then we could have 2-pin stacks and slice the symbol width roughly in half.

@Ratfink 
You wrote the generator for these symbols, Any comments?

evanshultz, 2018-09-22:
Oh, and the ground pins can always be stacked so that is a good and valid point.

DanSGiesbrecht, 2018-09-22:
@evanshultz thank you for the quick reply. I agree with having separate VDD pins for decoupling caps to indicate placement. Leaving all VDD pins separate (no stacks or pairs whatsoever) actually sounds good to me - allowing the designer to make the call on which pins should have decoupling caps based on their own judgement and design constraints.

For the GND side, I think that creating one VSS pin stack and separate analog VSSA pin(s) sounds like a good idea.

Ratfink, 2018-09-22:
Perhaps some of these power pins should be stacked, but I don't think there's anything in the XML source files from which these symbols are generated that indicates which ones should be.  In this case, we'd have to either (1) figure it out from the datasheet for each STM32 chip, or (2) play it safe and not stack any, as we've done.  Though as you point out, we could also (3) change our stacking rules to allow stacking ground but not power pins.

We can't change the symbol size until 6.0 though (I think? I'm only semi-in-the-loop since starting grad school), so I can't imagine there's much hurry to address this.

evanshultz, 2018-09-23:
Our stacking rules already require a single ground pin. This was from the original discussion (screenshots instead of links).

Ground pins fall into this statement:
[image]

And this followup covers partial stacking situations:
[image]

Some polish could be put on the wording for a KLC update, but AFAIK this is the current paradigm and how it applies here.

@DanSGiesbrecht 
If you're interested in this topic, stacking GND pins should be easy and won't break existing symbols (there will be little tendrils of wires sticking up, but also note how recently these symbols were merged so likely few users are using them now.

You could figure out which power pins or pin pairs could stack because those may or may not break a schematic. Either way, the script likely would need to learn how to deal with stacking power pins in a user-configurable way.

Splitting symbols can't happen yet as mentioned above, but you could also allow figure this out and allow a split/no-split switch in the script that we could throw later.

@Ratfink 
Congrats on grad school! I hope you're off to a good start!