KiCad/kicad-symbols #1031: Travis check doesn't rebuild

DanSGiesbrecht, 2018-10-04:
I noticed that currently, Travis doesn't rebuild/re-check if there are no commits to the symbols PR. This is problematic if a PR is waiting **only on a new footprint**. The Travis check will appear to still be failing (because it has not rebuilt), even if an associated footprint PR has been merged. This requires a librarian to periodically rebuild the Travis check manually to see if it is passing.

I have added the "Pending footprint" label to indicate that the Travis check should be rebuilt periodically, but does anyone know of a way to fix this? Can Travis be told to rebuild on a schedule, or be triggered by an external PR? I'm a bit green at this, so help is appreciated.

evanshultz, 2018-10-04:
I don't know of a way to periodically have it run, and that seems like a waste of resources.

Triggering on an associated PR might work, but I don't know how to do it.

Fortunately, footprint PRs are linked most of the time since we have such a good community of contributors. That means I can see in the symbol PR body the footprint PR link and its status so I don't have to leave to symbol PR to know I need to manually ask Travis to take a look.

Automation would be nice. @SchrodingersGat set things up initially and I believe @poeschlr may have done a bit of work with Travis also.

evanshultz, 2018-11-23:
I'm assuming this is the closable now. If somebody figures out a way to trigger Travis when a commit is pushed to a linked PR, then create a new issue or reopen.

closed 2018-11-23 by evanshultz