KiCad/kicad-symbols #1037: mcp1525 pin numbering

alpinekid, 2018-10-06:
In library: **Reference_voltage**
part MCP1525-TO; alias MCP1541-TO has a bad pin out.
There is a difference between the to-92 package and the sot-23-3 package.
The manufacture reversed the usage of pin numbers:-(

for the to-92 package,
Vss and Vin have been swapped between the to-92 and the sot.
Vss should be on pin 1, its the ground pin, on the to-92 
Vin should be on pin 3, its the supply voltage.

Its the reverse from the surface mount version of this part so it was an easy miss.
The mcp1541-TT is correct for surface mount package.
The mcp1541-TO symbol has the wrong pin outs.

The mfg part number difference is: through hole mcp1541 and surface mount is mcp1541T.
Its the suffix T that is important.

NB: its not a footprint issue, the pin number definition change between packages.

data sheet http://ww1.microchip.com/downloads/en/DeviceDoc/21653C.pdf

I know this because I just let the smoke out of a part.

evanshultz, 2018-10-06:
Pardon me for any rehashing, but let me dump all information I reviewed into this comment for verification.

Datasheet pinouts:
[image]

Indeed, I see the symbols for each package have the same pin order:
[image]

Since the footprint pinout could vary, let's confirm that too.

SOT-23:
[image]

TO-92_Inline:
[image]

Yep, the TO92 symbol needs pins1 and 3 reversed. I'm sorry that this error exists in the library and that it gave you a nasty surprise. However, thanks for the patience to report it.

Are you willing to submit a PR to fix the TO92 symbol's pinout?

evanshultz, 2018-10-08:
Fix submitted at https://github.com/KiCad/kicad-symbols/pull/1044.

poeschlr, 2018-10-08:
Fix merged

closed 2018-10-08 by poeschlr