KiCad/kicad-symbols #1040: KLC: Stacking of power pins

FabianInostroza, 2018-10-06:
A friend made a schematic and added an ATmega16A, this part has 3 VDD pins stacked and he added only one bypass capacitor. 
Would it be better to avoid stacking of power pins when the number of them is not high (not FPGAs nor large MCUs)?, this may prevent errors like this and also make the library consistent since other MCU libraries doesn't use stacking on power pins (LPC and STM32 at least).

PD: I'm aware that nothing beats reading the datasheet or that the _error_ would be detected in the layout phase.

evanshultz, 2018-10-07:
Sorry to hear about this and thank you for taking the time to post this message.

Our policy is to not stack power input (where the part draws power) pins if the datasheet calls out decoupling capacitors for each pin or gives other explicit reasons to keep the pins separate. For ground pins or other pins which do not have explicit instructions about any per-pin requirements, we stack pins.

Because the library is joined with the KiCad releases in an attempt to avoid breaking existing schematics unnecessarily, and also because some symbols are simply old, there are sure to be many symbols which violate the current guideline I mentioned above. Or the guidelines above may explain why some MCUs have stacked power pins and other don't.

Looking at the part datasheet (http://ww1.microchip.com/downloads/en/DeviceDoc/Atmel-8154-8-bit-AVR-ATmega16A_Datasheet.pdf), I just scrolled through it front-to-back and searched for a few key terms and I don't see anything about requiring a bypass cap for each VCC pin. Can you point me to the section of the datasheet or other documentation showing this requirement?

Since you used the word error in italic it makes me think this is a clear requirement from Atmel or just that you feel very strongly that not having a cap on each VCC pin is unacceptable. If there isn't any documentation requiring one cap for each VCC pin, then I assume that's your or your friend's "best practice" and so it's just what you do even if Atmel doesn't require it? In that case, it's hard to make an argument because a single cap might be perfectly fine in some or all applications, or perhaps two, or perhaps 3 are really required but not written down from Atmel. I don't honestly know, but if Atmel doesn't tell us what to do with VCC caps it falls into preference and applications-specific needs.

This has been discussed several times and there are valid reason for stacking and not stacking. We do expect the situation to be much better with an update to the symbol file format in a future KiCad release, but that doesn't change things now.

Thank you for your input since it does help guide how the official KiCad library is built. We are all volunteers trying to develop and maintain a powerful and useful library to support all KiCad users, like you and your friend.

FabianInostroza, 2018-10-07:
Hi @evanshultz 

I don't add decoupling capacitors in the power pins of digital circuits because the datasheet says so, I think that is best practice to reduce EMI and to have a good power supply. However I found an appnote that recommends adding decoupling capacitors to every pair of power pins [1] (see end of sec 2.1).

Now I think it was a mistake to use "error" in italics, quotation would have been better. What I wanted to say is that it may be perceived as an error for some people but not for others since the circuit may work fine most of the time, there could be unexpected resets, susceptibility to EMI or unintended radiations.

If the library maintainers agree with this I offer myself to do the modifications when the window to such modifications is open.

[1] http://ww1.microchip.com/downloads/en/AppNotes/00002519A.pdf

diegoherranz, 2018-10-24:
>  We do expect the situation to be much better with an update to the symbol file format in a future KiCad release

If a schematic pin could map to multiple layout pins, I think this problem would be gone and we would have the best of both worlds.
We could do pin stacking and you would still see that for instance VDD is mapped to a couple of pins on the screenshot below, so you may add more decoupling caps or whatever you consider.

[image]

If it's many pins, it wouldn't fit. But we could say something like "12 pins" and if you hover over the pin you see the full list of pins.

Just an idea.

Thanks.

antoniovazquezblanco, 2018-10-24:
Can we agree that we should unstack ATmega16A VCC pins? What do you @KiCad/librarians think?

evanshultz, 2018-10-26:
@diegoherranz 
Yes, had a similar thought some time ago, even as a stopgap with just text, but my suggestion didn't catch on. I would think something like you propose would satisfy all users and hopefully something like this is what we can expect to be implemented in KiCad.

@antoniovazquezblanco 
AN2519 is ambiguous and different ICs and designs will have different requirements. I would be happier with clear part-by-part information by Microchip that could be coupled with designer intuition/experience and testing. That being said, I'm fine with unstacking or only doing dual stacks on ATmega16A (and other?) symbols once we're on the 5.1 release cycle (I believe then is the time to make these kind of changes.)