KiCad/kicad-symbols #1065: No Footprints showing up in CvPCB

gabeart10, 2018-10-13:
When I went into CvPCB to assign my components to their footprints, there were no footprints found when I clicked on a component. I know it it not filtering and all my libraries seem to be good, so I am at a loss. I am pretty new to Kicad so I may be just doing something really dumb. 
[image]
[image]

gabeart10, 2018-10-13:
Also here is my Kicad version and OS
[image]

gabeart10, 2018-10-13:
Figured it out, Had to delate fp-lib-table and have it recreate it.

closed 2018-10-13 by gabeart10

poeschlr, 2018-10-14:
For the future: This is not the right place to ask user support questions. For that please head over to the forum: https://forum.kicad.info/