KiCad/kicad-symbols #1069: Git assistance with symbols

gcormier, 2018-10-15:
Wondering if I can get some Git assistance.

I recently contributed #1046 which was accepted.

During that time, I also started work on MICRF220. So in that branch, MICRF112 does not yet exist. I would like to fix this. From my understanding, 2 options

1) git rebase. But this might be a bad practice since I had pushed some commits to git already on the 220 brach.

2) git merge with the new master. This felt like the way, but I get these conflicts:
[image]

So, I could manually fix up the missing ENDDRAW/ENDREF, but I figured I would touch base with those perhaps regularly running in to this on how they handle it.

evanshultz, 2018-10-16:
I have done it manually with success in most situations. Sometimes Git, but more often GitHub, gets confused. Blowing away the branch and starting over is another option if things look good but you're not able to submit a correct PR from the original branch.

gcormier, 2018-11-20:
I ended up doing it manually as @evanshultz suggested. It seems a bit ghetto, but it's the path of least resistance. I don't know enough about git to know if there's a way to deal with merges like this, or if it would have to be a change in the library file to somehow have enough unique-ness that merge would pickup the new change easily.

closed 2018-11-20 by gcormier